Opentrons/opentrons #7364: spike: Research  / prototype protocol file runner

mcous, 2021-02-19:
## overview

As the Protocol Engine and control of protocol's via HTTP projects progress, we're going to get into a world where:

- We have a protocol file resources on the server thanks to HTTP protocol upload
- We have a protocol engine that controls protocol run resources that:
    - Needs to be accessible to the HTTP server so it can represent protocol resources
    - Needs to be accessible to a Protocol Context or JSON protocol runner in order to take protocol actions
- We have a need to pause, resume, or cancel the run of a protocol file

We will need something in place to thread these things together. 

## Existing functionality

Currently, this functionality is spread across two components, mainly:

- The RPC protocol session, which pauses, resumes, etc. by interacting with the hardware controller and allowing protocols to stop executing as a side-effect of that interaction
- A function called `run_protocol` that takes a protocol file resource and a `ProtocolContext` and blindly runs through the whole file
    -`api/src/opentrons/protocols/execution/execute.py`
    - "Runs through" via `exec`'ing a Python string or by for-looping through all the steps of a JSON protocol

## Existing research and thinking

- Python protocols can be executed in a debugger environment for better control of execution
- JSON protocols should not be run through the ProtocolContext

## Spike plan

Time-boxed to 5 points.

Test-drive through new protocol runner(s) from the top down. The goals of the spike are:

- Identify areas in the Protocol Engine that will require work to support using it for actual protocols
- Research architectural design for a system component in charge of protocol flow control
- Identify existing code and systems that can be adapted

Roughly, the runner(s) may look like:

- Dependencies
    - `ProtocolEngine` instance
    - (Probably) `HardwareController` instance
- Methods
    -  `initialize` - Takes an input protocol resource and sets up run state
        - For Python protocols, this might set up an execution environment of some sort
        - For JSON protocols, this may translate and load equipment and commands into the Protocol Engine
    - `run` - Starts a protocol run
        - For Python, start the Python execution in the debugger
        - For JSON, tell the engine to start processing queued commands
    - `pause` - Pauses a protocol run
        - Tell the Engine to avoid executing the next command
        - (Potentialy) Pause the HardwareController
        - (Potentially) tell the Python debugger halt at a given line
    - `cancel` - Cancels the protocol run
        - Tell the Engine to stop and update resource states accordingly
        - Issue hard halts on HardwareController resources
        - Tell the Python debugger to break execution

amitlissack, 2021-03-16:
Something that complicates the play/pause/cancel behavior is the various background tasks created in the Module derived classes.

amitlissack, 2021-03-17:
We should consider whether we want the ProtocolRunner to support pre-ProtocolEngine context implementation.

closed 2021-04-20 by mattwelch