Opentrons/opentrons #7424: Protocol Engine tip tracking (WellImplementation.set_has_tip)

amitlissack, 2021-03-04:


mcous, 2021-03-15:
A few notes from taking a read through existing tip tracking logic, which is spread between the following files, currently:

- `opentrons.protocol_api.InstrumentContext`
- `opentrons.protocol_api.Labware`
- `opentrons.protocols.context.protocol_api.Labware`
- `opentrons.protocols.api_support.TipTracker`

From what I can tell, we need to support the following behaviors in the execution engine:

- Tip pickup
    - Result: pipette has a tip attached
    - Result: well no longer contains a tip
    - Potential failure: pipette already has a tip attached
    - Potential failure: labware is not a tiprack
    - Potential failure: well has no tip
    - Potential failure: well has dirty tip and dirty tip pickup not allowed (PAPI version > 2.2)
- Tip drop
    - Result: pipette no longer has a tip attached
    - Result: well now contains a dirty tip
    - Potential failure: labware is not a tiprack nor trash container
        - Note: how do we specify custom trash?
    - Potential failure: well already has a tip in it

Additionally, multi-channel pipettes add some wrinkles:

- A multi-channel pipette can pick up anywhere between 1 and 8 tips
- A multi-channel pipette needs several clear "wells" to drop tips

A sort of big question falls out of this, which is **how do we specify well access for multi-channel pipettes in the Protocol Engine?** The answer to this question is important to ensure we track all necessary tip state. It also has implications for any future liquid state. Kicking off this discussion internally and with SPDDRS to help figure out this ticket

mattwelch, 2021-09-07:
Closing due to http/experience-led protocol engine story/roadmap route taken.

closed 2021-09-07 by mattwelch