Opentrons/opentrons #7459: Batch Edit Mix Forms

anthonywilleee, 2021-03-08:
**User Story**
As a Protocol Designer user, I would like the ability to batch edit Mix steps, so that I can speed up my protocol design process.

**Description**
This epic is a fast follow to Batch Edit MVP, where we introduced multi-select and batch edit Transfer forms functionality.

Similar to batch edit Transfer forms, enabled advanced settings for Mix steps will be batch editable (see advanced settings enabled conditions [here](https://docs.google.com/spreadsheets/d/1dbEeeFj3Bdi9IsRQUl_N-xUr3XnERoJkL2BhEuWO9gw/edit#gid=0)

**Requirements**

- All advanced settings for Mix are batch editable across Mix steps, as long as requirements to enable changes are met (e.g., pipette type must be the same)

- Prevent users from accidentally making changes to batches of steps

- Save button displays and confirmation that save has been applied to selected steps

- Display editable advanced settings that share values, and display a mixed state "-" for values that differ between steps

- All modal considerations from Transfer steps apply

anthonywilleee, 2021-03-16:
when we start this, consider the tip postion troubles

IanLondon, 2021-03-30:
> when we start this, consider the tip postion troubles

#7470 for reference 🙃

closed 2021-06-07 by anthonywilleee