Opentrons/opentrons #7466: Ensure tip is always dropped after run ends

sfoster1, 2021-03-10:
If a protocol does not end with a drop tip command, or if a protocol is cancelled partway through due to user action or an internal error, a tip that may have liquid will be attached to a pipette when it homes its plunger. This results in two things: a) the home action could overaspirate and get liquid inside the pipette body; b) the tip will stay on there and the user will have to take it off manually.

Instead, when a protocol ends the robot should
- home z/a
- home x/y
- do a drop tip
- home b/c

This will ensure that the tips are dropped in the trash along with any liquid they contain before that liquid is overaspirated.

SyntaxColoring, 2021-03-12:
Related / maybe dupes?

* #3738
* #5428

mattwelch, 2021-07-13:
@nusrat813 we have brought this in to sprint 37. when does this work need to be merged to make 4.5 (July 28).

nusrat813, 2021-07-19:
@mattwelch we would need this to be merged no later then July 26 if we want a release on July 30th

mcous, 2021-07-27:
Revising estimate from 3 to 5 due to discussions with the embedded team. It's hard

PR #8229 (2021-08-30): fix(api): drop tip after cancel
Closes #7466
Closes #3738
Closes #5428

# Overview

This PR adds a functionality that if a protocol is cancelled mid-run and the pipette has a tip attached when the cancel is initiated, then the robot will empty the tip & drop it in the pipette's trash container before homing the plungers.

This is achieved in the following way:

Right after a cancel is issued:
1. pipette state is stored to determine whether it has a tip (and to save the last known plunger location but I’m not using that right now)
2. Cancel is fully executed with `home_after=False`. Hardware co [... cut]
closed 2021-08-30 by sanni-t