Opentrons/opentrons #7477: Spike: Bug compatibility in Python Protocol API versioning

SyntaxColoring, 2021-03-12:
* The Python Protocol API has a large surface area of undefined behavior
  * Partially due to lack of parameter validation
* The API sometimes also has bugs
* I feel it's unduly difficult to fix these bugs because of our versioning scheme: the current version of the software must try to be bug-for-bug compatible with all prior versions of the software
* Fully characterizing the behavior of existing bugs is laborious and error-prone

What can/should we do about this?

# Deliverables

Document describing:

* Background (what problems do we feel exist, and why are they bad)
  * Concrete examples
  * We want to make it easier to develop correct protocols without scrutinizing a wall-of-text run log or watching the robot move
* Non-exhaustive list of directions for solutions (at a high level)
* Prior art?

## Purpose

Aligning, Software-wide (or at least CPX-wide), on what to do about this.

* Doing nothing is a valid answer.
* Policy-level solutions are a valid answer (e.g. loosen our versioning policy).
* Technical solutions are a valid answer (e.g. multiple versions of the Python API installed concurrently, via virtualenvs or whatever).

Implications for Protocol Engine?

## Audience

Depending on the solutions that are explored:

* All robotics platform software developers
* Product (i.e. @mattwelch)
* Applications Engineering
* Support maybe (the high-level parts)

SyntaxColoring, 2021-03-12:
Probably has implications with Protocol Engine...

* We suspect PE would warrant a major software version bump anyway...maybe this is an opportunity to batch up "breaking" "changes" to the Python API like bug fixes like #6754?
* Still, our bug-compatibility versioning policy will still be a problem in PE. How do we deal with that? We can "reset ourselves" with a 4.0->5.0 version bump, but the hairs will still eventually grow as long as we have our current versioning policy. (They'll just grow in different places.)

Mitigation strategies?

* Make exhaustive unhappy-path tests a hard requirement of PE?
* Make very strict parameter validation a hard requirement of PE?

Other things PE should consider?

* Facilitation of non-fatal warnings from PE?
  * Source mapping (step or line of code)?
  * Don't necessarily want it to be part of the run log

SyntaxColoring, 2021-03-17:
Messy thoughts.

We had a few motivations for giving Python protocols mandatory version metadata. The motivations are related to each other, but distinct.

1. Safely changing default behavior. For example, if `touch_tip` works one way today and a different way tomorrow, people's protocols shouldn't behave differently when they update their robots.
2. Helpful error messages when, because a protocol uses new fea a protocol can't work on an out-of-date robot. For example, if we add a new heater-shaker module...to use the new module, you need to edit the file to use a newer `apiLevel`. 
3. The above, but for bugfixes instead of features.

I think (1) is uncontroversial. I'm less sure about (2) and (3).

Suppose someone writes a protocol that uses a new `fancy_transfer()` feature on a new robot, and then they share it with their friends who might be on old robots.

With (2) and (3), we have this:

| | Protocol requests modern `apiLevel` | Protocol requests old `apiLevel` |
|-|-|-|
| **Robot supports modern `apiLevel`** | The protocol is runnable. | Specific, actionable, easy-to-resolve, high-level error: "Use an apiLevel >= 2.10 to use fancy_transfer()." |
| **Robot only supports older `apiLevel`s** | Specific, actionable, high-level error: "Update your robot software to run this protocol." This *should* be easy to resolve, but users might require revalidation. | Low-level error: "No such method exists." Users might need help from Support. |

Without (2) and (3):

* The top-right cell would change to "The protocol is runnable."
* The bottom-right cell would happen more.
* Things might be more maintainable for us, because we wouldn't have to carry around as much old code?

Questions:

* We should check Support cases and the analytics on error messages. How many people are running into the bottom-left cell today?
* Would removing (2) and (3) actually be a win for maintainability?
* In this formulation, where's the line between "bug fix" and "changing default behavior"?

SyntaxColoring, 2021-03-18:
For near-term Protocol Engine work, the part of this that's especially relevant is:

* Clarifying what problems we think exist in the status quo (and making sure we all agree that those problems are problems)
* Coming up with a rough vision for how Protocol Engine will address those concerns (including hypothesizing early traps to avoid)

mattwelch, 2021-06-15:
Closing in favor of a smaller follow up story that @SyntaxColoring will write for next sprint.

closed 2021-06-15 by mattwelch