Opentrons/opentrons #7499: [bug] After switching to other pipette, current pipette fails to home if other was just homed.

gbohner, 2021-03-16:
I'm using the Jupyter notebook to test movement commands on API v2.9.

The following code creates unexpected (at least to me) and dangerous behaviour, that is the previously used pipette stays at its current Z height instead of homing before movement.

```python
import opentrons.execute
from opentrons.types import Point, Location
protocol_context = opentrons.execute.get_protocol_api('2.9')

pipette_20 = protocol_context.load_instrument(
        "p20_single_gen2", mount='left', tip_racks=[])
pipette_300 = protocol_context.load_instrument(
        "p300_single_gen2", mount='right', tip_racks=[])

protocol_context.home()
pipette_20.move_to(Location(Point(224,230,139.9), None), speed=100, minimum_z_height=130)

# this line is causing the erroneous behaviour during the next line
pipette_300.home() 

# This line normally would first home the "20 ul pipette" before moving, however due to the .home() previously, it forgets to do so
pipette_300.move_to(Location(Point(224,230,139.9), None), speed=100, minimum_z_height=130)
```

At the end of this, both pipettes are at a "low" position, which as far as I understand should never happen.

Server version: 4.1.1
Firmware Version: v1.1.0-25e5cea

As a related feature request, it would be great if during normal behaviour, instead of fully "homing" the "other" pipette (which takes forever to check with the limit switches), it would simply raise it - I understand it's better to check everything from hardware instead of software, but it would be great to have a "dangerous" mode for people who like to go fast :)

SyntaxColoring, 2021-03-16:
Thanks for reporting this! I'm labeling this ticket for our own internal tracking purposes, but we haven't taken a close look at this yet.

gbohner, 2021-03-16:
After playing around a bit more, I think the reason for the bug is that the "instrument_context.home()" command likely sets a "homed" flag for both instruments (instead of just the one), and instruments that are already homed (correctly) do not need to re-home before another instrument is being used.

SyntaxColoring, 2022-07-07:
Confirmed that this is still a problem on v6.0.0-beta.1.

mcous, 2022-07-11:
Unintentionally closed

mcous, 2022-07-27:
Fix will be present in the upcoming `v6.1.0` release. You will need to update your protocol to use Protocol API version 2.13 to start using the corrected behavior

PR #11072 (2022-07-27): fix(api): if needed, home both pipettes with InstrumentContext.home
## Overview

When using `HardwareControlAPI.home_z` (by way of `InstrumentContext.home`), the HW API can be put into a state where it thinks both pipettes are retracted, when in reality, only one pipette is retracted. In this state, movements become very likely to cause collisions.

Fixes #7499

## Changelog

Now, when calling `InstrumentContext.home`, 

- fix(api): if needed, home both pipettes with InstrumentContext.home

## Review requests

I tested the following behaviors on an OT-2R in the office. I think given these smok [... cut]
commit 0115066c (2022-07-27): fix(api): if needed, home both pipettes with InstrumentContext.home (#11072)

Fixes #7499
closed 2022-07-27 by mcous