Opentrons/opentrons #7508: bug: PD - touch tip minus button reduces the value below the minimum range

reachojhaneha, 2021-03-17:
Edit the tip position custom settings with minus sign. It reduces the value below the minimum range, this is a regression introduced. Reducing the value of tip position using minus button  below minimum value is not possible in Production PD.

IanLondon, 2021-03-17:
I think this one is desired behavior, because in prod (old Tip Position modal) we didn't allow out-of-range values, which was annoying bc you couldn't type freely. Now we do allow values even when out of range. The pattern of allowing any number to be entered through typing is continue in allowing any height to be set with the arrow keys.

reachojhaneha, 2021-03-17:
I just checked with Morgan about this and we concluded that we should preserve the "minus button" lowering the value to minimum value and not below minimum. Because the minus button does not lower the value to "0". For example if the minimum value for tip position is 12.5, minus button if clicked multiple times lowers the value till 12 and not 0.

IanLondon, 2021-03-17:
Could we rename + relabel this as a feature and not a bug? Was not specified in the ticket #7143 when this was done

anthonywilleee, 2021-03-30:
@howisthisnamenottakenyet - also curious your thoughts on this one..should we preserve not being able to go past min / max when hitting minus button

PR #7592 (2021-04-07): fix(protocol-designer): fix tip position modal minus button
# Overview

Closes #7508

# Changelog

# Review requests

See #7508

# Risk assessment

Low
commit bb93c7ce (2021-04-08): fix(protocol-designer): fix tip position modal minus button (#7592)

Closes #7508
closed 2021-04-07 by IanLondon