Opentrons/opentrons #7510: Is it possible change `tip_use_strategy` during operation?

partrita, 2021-03-18:
```python
def get_values(*names):
    import json
    _all_values = json.loads("""{"pipette_type":"p300_multi","dilution_factor":2,"num_of_dilutions":10,"total_mixing_volume":200,"tip_use_strategy":"always"}""")
    return [_all_values[n] for n in names]

def run(protocol_context):
    [pipette_type, dilution_factor, num_of_dilutions, total_mixing_volume,
        tip_use_strategy] = get_values(  # noqa: F821
            'pipette_type', 'dilution_factor', 'num_of_dilutions',
            'total_mixing_volume', 'tip_use_strategy'
        )
    transfer_volume = total_mixing_volume/dilution_factor
    diluent_volume = total_mixing_volume - transfer_volume
```

Is it possible change `tip_use_strategy` during operation?
For example, to serial dilution i want to use one tip for filling `transfer_volume` for 96 wells and exchange a tip for `diluent_volume` fill.

SyntaxColoring, 2021-03-23:
Hey @partrita. 👋 

That looks like a protocol that came from https://protocols.opentrons.com/. Is that where you downloaded it? Are you comfortable writing Python yourself?

If you're comfortable writing Python yourself, then yes—you can choose when the OT-2 changes tips, but you might need to split a big `transfer()`, `distribute()`, or `consolidate()` command into multiple smaller ones. See, for example, the [`new_tip` kwarg for `InstrumentContext.transfer()`](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.InstrumentContext.transfer).

partrita, 2021-04-11:
@SyntaxColoring Thank you for answer.

Yes. I know how to code python. but I didn't know it was in the API documentation.
I will take the time to read more.

closed 2021-04-11 by partrita