Opentrons/opentrons #7521: bug: touch_tip too high in distribute

dleopold, 2021-03-19:
# Overview

When using `touch_tip=True` in the distribute command the tip is too high. I have double checked all off the calibrations multiple times, but the is always ~5mm above the well when the program attempts to touch the sides of the well. As a result, if there is a drop clinging to the tip is can end up in a neighboring well. I have tried this on multiple plate / tube type and always observe the same behavior. This is my first attempt at working with out new OT-2, so I could be making a simple error, but I can not seem to explain why this is happening. Everything except the `touch_tip` behavior is working as expected.

# Steps to reproduce

This is the code causing the error:

```
pipette300.distribute(24, mastermix.wells()[0], [destination.wells()[well] for well in range(48)], 
        blow_out=True, blowout_location='source well', disposal_volume=12, touch_tip=True)

```
# Current behavior

Described above.

# Expected behavior

I expect the tip to be slightly below the top if the well when the robot moves to touch the tip against the sides.

SyntaxColoring, 2021-03-23:
Hey @dleopold, thanks for reaching out, and sorry this is your first experience with your new OT-2!

If you do:

```python
pipete300.move_to(destination.wells()[0].top())
```

Is the bottom of the tip even with the top of the well, or is it ~5 mm too high like you observed during the `touch_tip`?

dleopold, 2021-03-26:
Sorry for the delayed reply. I am not in the lab every day and I wanted to wait until I could trouble shoot a bit more. The issue appears to be with z-axis calibration of the 300 ul (GEN2) pipette. Despite re-calibrating multiple times (I even physically removed and replaced it) the pipette always stops ~2-5 mm above the well of any plate/tube. The issue does not seem to apply to the p20 also loaded at the same time.

For example here are the steps I just followed:
1. Using the opentrons app, I run through the calibration steps from the beginning for both the p300 and p20
2. From a ipython notebook I load a rack of 20ul tips, 300ul tips and a 96 well plate
3. For the p20 - I pick up a tip and move to the top of the first well and the tip position is correct.
4. For the p300 - I pick up a tip and move to the top of a well and the tip is actually ~4-5mm above the top of the well.

I have repeated this after recalibrating multiple times and with different plates and tube and the p300 always has the wrong z-axis calibration. I have also tried with 300ul and 200ul filtered tips on the p300 and the behavour is alwasy the same.

If I load a protocol that uses both pipettes into the app and confirm the calibration (i.e. "proceed to calibration") I can correct the Z-axis for the p300 for a specific plate, but then the z-axis calibration for the p20 will be off for that plate and it will bend the tip against the bottom of the well!

I am out of ideas. Any help figuring out why the z-axis calibration for the p300 is not working right would be greatly appreciated.

SyntaxColoring, 2021-03-26:
Hmm. Could you try this after calibrating, and before running anything in Jupyter Notebook (IPython):

1. In Jupyter Notebook, select **Kernel > Restart & Clear Output** from the menu bar.
2. Wait a few seconds for the notebook to reset. You should see a little blue "Kernel ready" message near the top-right.
3. Run your code again and see if the calibration is correct this time.

dleopold, 2021-03-26:
Nope. The z-axis is still ~+5mm from the top of the plate after recalibrating (using the app, not sure if calibration is possible through the Jupyter Notebook) and selecting the Kernel > Restart & Clear Output.

SyntaxColoring, 2021-03-26:
Drat.

If you run your protocol through the Opentrons App instead of Jupyter Notebook, does the calibration seem correct, then? (Testing this might take some restructuring of your Python, since the Opentrons App expects protocols in [a different structure](https://docs.opentrons.com/v2/index.html#how-it-looks) than what's valid in Jupyter Notebook.)

> Not sure if calibration is possible through the Jupyter Notebook

It's not, currently.

dleopold, 2021-03-26:
No. The same issue occurs if I upload a python protocol in the app.

SyntaxColoring, 2021-03-26:
Hm, okay. So it sounds like this is really an issue with your OT-2's calibration in general—it's not a bug specific to `touch_tip` or Jupyter Notebook.

This is straying outside my expertise, so I'm asking internally what our next steps should be.

jepdavidson, 2021-03-27:
Hi dleopold,

I have a support ticket open that I raised after commenting on issue [7139](https://github.com/Opentrons/opentrons/issues/7139).  I actually referenced your issue in my support email as it seemed very similar to what I was seeing, and they agreed that the issues are likely related.

Anyway, I have been advised to run a calibrations reset from the Opentrons App, then re-do the calibrations and see if things work as expected after this.  Support have said that this procedure has helped resolve this "bug" in 4.2.0 for others.

I hope to try this on Monday, but wanted to post here in case this helped you out.

Kind regards

James

SyntaxColoring, 2021-03-29:
@dleopold: Our Support team asks that you email them at support@opentrons.com and mention that you're @dleopold in this GitHub issue (#7521). They'll be able to troubleshoot some more with you, and they should also have context to correlate your symptoms with symptoms reported by other users to narrow down where the problem is.

I'm closing this issue because it doesn't look like there's a bug where `touch_tip` is too high in a `distribute`. But it definitely sounds like there's something fishy going on, and our Support team will work with you to keep looking into it. Thanks again for reaching out!

closed 2021-03-29 by SyntaxColoring