Opentrons/opentrons #7531: feat: version 3 app allows "A0", version 4 app does not. Please allow "A0" again.

bmyury, 2021-03-23:
Hi, 

We are using custom labware definition for our device.
 
Our device is pretty finicky (calibration process has a potential to destroy it). To this end we assigned the "A0" position specifically for calibration. A0 is located separately from all other "wells" which keeps them safe yet during the protocol the reactions happen in wells A1....C4 (totall 12 wells).

It worked just fine in server and app version 3. But in version 4 it is not allowed (the regular expression for the well names does not allow for "A0" – the wells could only start from A1. I assume the old regex was:
'^([A-Z]+)([0-9][0-9]*)$' while the new is '^([A-Z]+)([1-9][0-9]*)$'

Could you please reverse this change and allow for the "A0" name of the well to be used again.
Thank you

Yury

mcous, 2021-03-23:
@bmyury can you expand a little bit on what "not allowed" means? What was the specific behavior you used in v3, and what is the behavior of v4 that is blocking you?

amitlissack, 2021-03-23:
This regex was added October of last year when I added row and column attributes to Well objects.

bmyury, 2021-03-23:
In v3 – I can use labware with "A0" in it. In v4 scripts with this labware definition throw an error saying that "A0" can;t be recognized by  regular expression.

below is my labware def

{"ordering":[["A0","A1","A2","A3","A4"],["B1","B2","B3","B4"],["C1","C2","C3","C4"]],"brand":{"brand":"Parhelia","brandId":["par2_cslp_metal_v1"]},"metadata":{"displayName":"par2_cslp_metal_v1","displayCategory":"wellPlate","displayVolumeUnits":"µL","tags":[]},"dimensions":{"xDimension":127.71,"yDimension":85.43,"zDimension":47.7},"wells":{"A0":{"depth":0,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":8.6,"y":75.7,"z":47.7},"A1":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":21.5,"y":70.5,"z":38.5},"B1":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":21.5,"y":51.3,"z":38.5},"C1":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":21.5,"y":32.5,"z":38.5},"A2":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":48.5,"y":70.5,"z":38.5},"B2":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":48.5,"y":51.3,"z":38.5},"C2":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":48.5,"y":32.5,"z":38.5},"A3":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":75.5,"y":70.5,"z":38.5},"B3":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":75.5,"y":51.3,"z":38.5},"C3":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":75.5,"y":32.5,"z":38.5},"A4":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":102.5,"y":70.5,"z":38.5},"B4":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":102.5,"y":51.3,"z":38.5},"C4":{"depth":9.2,"totalLiquidVolume":300,"shape":"circular","diameter":10,"x":102.5,"y":32.5,"z":38.5}},"groups":[{"metadata":{"displayName":"par2_cslp_metal_v1","displayCategory":"wellPlate","wellBottomShape":"flat"},"brand":{"brand":"Parhelia","brandId":["par2_cslp_metal_v1"]},"wells":["A0","A1","A2","A3","A4","B1","B2","B3","B4","C1","C2","C3","C4"]}],"parameters":{"format":"irregular","quirks":[],"isTiprack":false,"isMagneticModuleCompatible":false,"loadName":"par2_cslp_metal_v1"},"namespace":"custom_beta","version":1,"schemaVersion":2,"cornerOffsetFromSlot":{"x":0,"y":0,"z":0}}

[image]

bmyury, 2021-03-24:
Dear @mcous @amitlissack – if my suggestion us accepted, when realistically can I expect it to be implemented and incorporated into an update. Just to give us an idea – to what to tell our colleagues that use our device

mcous, 2021-03-25:
@bmyury this will make it into the `v4.3.0` release in about a month. In the mean time, does this (temproary!) workaround work for you? It seems to allow your definition to load for me:

```py
import re

metadata = {
  'apiLevel': '2.9'
}

def run(ctx):
    # HACK: patch _private internals_ of well logic to allow wells columns starting with `0`
    # not necessary for opentrons >= v4.3
    try:
        from opentrons.protocols.implementations.well import WellImplementation
        # replace row/column identification regex for wells
        WellImplementation.pattern = re.compile(r"^([A-Z]+)([0-9]+)$", re.X)
    except Exception as e:
        ctx.comment(f"Failed to patch WellImplementation: {e}")
```

bmyury, 2021-03-25:
Thank you so much @mcous – I will try it and will let you know

commit 1f6b11bd (2021-03-31): fix(api, shared-data): Regular expression used to split well into row and column did not allow a 0 column. (#7566)

closes #7531