Opentrons/opentrons #7533: Connecting to Opentrons using Windows 10/MAC gives "Invoke-Webreuqest: : {"error": "bad-interface", "message": "The endpoint http://127.0.0.1:34000/server/ssh_keys can only be used from local connections"}"

beniroquai, 2021-03-24:
# Overview
When following the Tutorial to connect to the opentrons [here](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2), I get the following message in a MAC terminal and Windows PowerShell:

```
PS C:\Users\diederichbenedict\Downloads> @{key = Get-Content ot2_ssh_key.pub | Out-String} | ConvertTo-Json | Invoke-WebRequest -Method Post -ContentType 'application/json' -Uri 192.168.43.215:31950/server/ssh_keys
Invoke-WebRequest : {"error": "bad-interface", "message": "The endpoint http://127.0.0.1:34000/server/ssh_keys can
only be used from local connections"}
At line:1 char:70
+ ... rtTo-Json | Invoke-WebRequest -Method Post -ContentType 'application/ ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-WebRequest], WebExc
   eption
    + FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand
```
# Steps to reproduce

Follow the tutorial [in ](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2) and execute Step 3

The opentrons and both computers are in the same network (WiFi). Connecting to it using SSH gives:

```
PS C:\Users\diederichbenedict\Downloads> ssh root@192.168.43.215
root@192.168.43.215: Permission denied (publickey).
```

# Current behavior

Error (see above)

# Expected behavior

Sharing public keys.

SyntaxColoring, 2021-03-24:
Hey @beniroquai, thanks for the report.

I think this is because you're connecting to your OT-2 via Wi-Fi, instead of via a cable. For security, we only accept new SSH keys when there's a direct wired connection between the computer and the OT-2 (a "link-local" connection), not a wireless connection. This is meant to prove that the person trying to add the SSH key has physical access to the OT-2.

This is mentioned in [the tutorial you linked](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2):

> To show that you have physical access to the OT-2, **you must connect to it via USB for this part.** A Wi-Fi connection will not work.

Though it looks like we could make that part a little harder to miss, and the error message a little clearer.

beniroquai, 2021-03-25:
Ah, I see. It's always the user's fault ;-)
Once I generated the ssh-key, I could connect to the opentrons through
WiFi?

Is there by any means a possibility to connect to the opentrons using plain
`ssh root@ROBOT_IP`

Thanks!

mcous, 2021-03-25:
Hi @beniroquai, that is correct: once the SSH key is generated and uploaded to the robot over the **wired** connection, you may use **any** connection - wired or Wi-Fi - to connect to your OT-2 via SSH.

> Is there by any means a possibility to connect to the opentrons using plain `ssh root@ROBOT_IP`

Sometimes this works well, and sometimes it doesn't! Depending on your operating system / networking setup, you may _also_ be able to connect with `ssh root@${robot_name}.local` where you replace `${robot_name}` with the name that appears in the Opentrons App.

If you are able to assign a static IP address on the Wi-Fi network to the OT-2, that's usually going to give you better results with `ssh root@${robot_ip}`.

SyntaxColoring, 2021-03-25:
Yeah, using a `.local` hostname is a good way to avoid having to find and copy-paste the robot's IP address.

To also avoid having to pass `-i ~/ot2_ssh_key`, you might be able to reconfigure your SSH tool. On macOS, here's a snippet from my `~/.ssh/config`:

```
Host my-robot
	Hostname ot2cep20190716a09.local
	IdentityFile ~/ot2_ssh_key
	User root
```

This lets me connect with just:

```console
ssh my-robot
```

beniroquai, 2021-03-26:
I see! Thanks for clarifying this!
A follow-up question, but perhaps a new issue: Is it possible to somehow do
"debugging" using visual studio code inside the opentrons?
Also: Is it possible to remote control the opentrons through ssh/http? (a
simple yes/no is enough ;-)

Thank!

>

mcous, 2021-03-26:
I'm going to close this issue because your problem has been resolved and we've created a ticket (#7538) to clarify the error message you received. Feel free to use this thread to continue asking questions, though!

> A follow-up question, but perhaps a new issue: Is it possible to somehow do
"debugging" using visual studio code inside the opentrons?

Unfortunately, no, not at the moment. This is a very cool idea that we'd love to do, but is not currently on our roadmap due to technical constraints and not having enough people to work on it.

> Also: Is it possible to remote control the opentrons through ssh/http? (a
simple yes/no is enough ;-)

Great question! 

**SSH**

At the moment, we have a [very basic utility called `opentrons_execute`](https://docs.opentrons.com/v2/new_advanced_running.html#command-line) for running Python protocols while SSH'd into the robot. 

**HTTP**

We are building an HTTP interface right now. Currently we have a beta program available that allows protocol file upload and run controls (pause, resume, cancel) via HTTP. Let me know if you're interested, and I can add you to the beta group!

Our intention is to eventually allow full protocol control over HTTP (e.g. tell the robot to aspirate via HTTP), but we're a ways off from that goal at the moment.

closed 2021-03-26 by mcous

beniroquai, 2021-03-26:
That sounds terrific! Especially the very last comment about **HTTP**-ing commands! Right now the robot is in control of devices in the network, but having the master in charge of multiple devices could lead to the easier formation of feedback loops e.g. by means of autonomous labs, where AI drives the flow. 
We are working on a small project, where a microscope captures images periodically; Having a feedback loop e.g. on the cell's state would already be awesome. The Raspberry Pi is a nice device, but not really well equipped with computational power. Cloud-based computing is of course an alternative. Let's see how far we can go. Further info [here](https://github.com/beniroquai/Opentrons-Microscope-Platereader)

Thanks for making this robot so awesome! Really cool thing!

beniroquai, 2021-04-04:
> **HTTP**
> 
> We are building an HTTP interface right now. Currently we have a beta program available that allows protocol file upload and run controls (pause, resume, cancel) via HTTP. Let me know if you're interested, and I can add you to the beta group!
> 

Would it be possible to get early access already? 

> Our intention is to eventually allow full protocol control over HTTP (e.g. tell the robot to aspirate via HTTP), but we're a ways off from that goal at the moment.

Cool! :) Do you think there is already a way to remote control the basic functions from a "Master" computer that performs the major computational and planning of the next experiments? Would I need to write a server that maps HTTP requests into opentrons-API commands running on the Raspberry Pi inside the OT-2? Are you aware of any work which does that already? Thanks!

mcous, 2021-04-06:
Hi @beniroquai, go ahead and shoot me an email at mike at opentrons dot com and I'll get you set up (and it might be easier to have a more broad discussion than a GitHub ticket thread!)