Opentrons/opentrons #7537: feat: protocol.run('script.sh') like command

pschupp, 2021-03-24:
It would be usefull to run small scripts (bash/python/etc) when a certain point in the protocol is reached by the robot. Specifically, I would love to get a ping (email/IRC/Slack/etc) from the robot when an incubation begins as our protocol involves transfer to a high-speed shaker.

mcous, 2021-03-25:
@pschupp if you're talking about Python protocols, are you able to do this using [vanilla Python](https://docs.python.org/3/library/subprocess.html#subprocess.run)?

```py
# ...
import subprocess
# ...
metadata = {
  'apiLevel': '2.9'
}
# ...
def run(ctx):
	# ...
	ctx.comment('Run myscript.sh')
	if not ctx.is_simulating():
		result = subprocess.run("/path/to/myscript.sh")
	# ...
```

...where `/path/to/myscript.sh` is a script that you've copied over to the robot filesystem via `scp` (or similar).

mattwelch, 2021-04-09:
We're presuming the above example solves the user's request. Please let us know if you have any further questions.

closed 2021-04-09 by mattwelch