Opentrons/opentrons #7590: bug: Y Offset on every robot movement while running a protocol as a service

lucas-j-provectusalgae, 2021-04-06:
# Overview

We're trying to run a protocol as a service on the OT-2. We're getting an offset that happen event after re-calibration of the robot. The robot can't pick-up any tips and the offset seems to be on the Y axis after experimenting on 2 different machines.
A simple protocol working when running as a python script doesn't work as a service.

# Steps to reproduce

**Correct operation :** 
(change the pipette and tiprack loaded)

Put a python script test_protocol.py in /data/ on the robot through scp :
```sh
import opentrons.execute
from opentrons import protocol_api
def trying():
    protocol = opentrons.execute.get_protocol_api('2.9')
    tiprack20 = protocol.load_labware('opentrons_96_tiprack_20ul', '5')
    pip20 = protocol.load_instrument('p20_single_gen2', 'right', tip_racks=[tiprack20])
    protocol.home()
    pip20.pick_up_tip()
    protocol.home()
    pip20.return_tip()
if __name__ == "__main__" :
    trying()
```

Run the protocol using : 
```sh
python3 /data/test_protocol.py
```
Everything should work fine.

**Wrong operation as a service :** 
Then to get the error, create a service file in /etc/systemd/system/
```sh
cd /etc/systemd/system/
mount -o remount, rw /
>test-protocol.service
```
Write the file using vim :

```sh
[Unit]
Description=test protocol service
After=network.target
StartLimitIntervalSec=0
[Service]
Type=simple
Restart=always
RestartSec=1
User=root
ExecStart=python3 /data/test_protocol.py
[Install]
WantedBy=multi-user.target
```

Then start the service using :

```sh
systemctl start test-protocol.service
```
You can see the robot arm offset. The robot can't pick-up the tip.

# Current behavior

The robot can't pick-up a tip most of the time. If it can pick up the tip both time, you still can see an offset that looks to be on the Y axis.

# Expected behavior

The protocols should run exactly the same without any offset.

mcous, 2021-04-06:
Hey @lucas-j-provectusalgae, this is a shot in the dark, but our system is really not designed to allow mounting `/` as read-write. I **highly** recommend that you **do not do this**; this partition is read-only to prevent accidents from getting the OS into an unbootable state.

I wonder if that is caused by sort of file permissions problem, perhaps due to an inability to load calibration data when run via `systemd` from `/etc`. Have you tried loading the unit with [`systemd-run`](https://www.freedesktop.org/software/systemd/man/systemd-run.html) instead? e.g.

```shell
systemd-run \
  --unit=test-protocol \
  --remain-after-exit \
  --setenv=OT_SMOOTHIE_ID=AMA \
  --setenv=RUNNING_ON_PI=1 \
  python /data/test-protocol.py
```

Off the top of my head, it's probably worth playing around with the `--working-directory` and/or `--user` options.

Finally, is there anything in the logs for your service that point towards any problems during run?

mikolaj-w-provectusalgae, 2021-04-30:
Hey @mcous, We resolved the offset issue. We didn't run our service with `RUNNING_ON_PI` env variable. Once added, everything works fine.

mcous, 2021-04-30:
@mikolaj-w-provectusalgae noted, that makes sense. I think if `RUNNING_ON_PI` is not set, a different calibration data directory is used, which explains the discrepancy. My recommendation still stands about avoiding `mount -o remount, rw /`.

Going to close this ticket since it seems like the issue has been resolved!

closed 2021-04-30 by mcous