Opentrons/opentrons #7591: feat: Add POST endpoint to aspirate/dispense/blowout/pickup/remove pipette possible?

beniroquai, 2021-04-06:
# Overview

In order to integrate the opentrons into our workflow which includes microscopic imaging and an image processing pipeline, it would be neat to control the opentrons and microscope from one place. Since the raspberry pi is cool, but does not have enough computational power to do the majority of the image processing part, I thought it would be great to have another computer as the "Master" in the workflow. I discovered that I can send HTTP requests to the robot in order to control it eg:

```py
headers = {'content-type': 'application/json',
           'opentrons-version': '*'}
path = 'http://21.3.2.5:31950/robot/home'
payload = {
"target": "robot",
"mount": "right"
}
x = requests.post(path, data = json.dumps(payload), headers=headers)

print(x.content)
```

My question would now be if it's also possible to aspirate/dispense/blow-out/pickup tips/remove tips using the fastAPI interface? I have not found any suitable endpoint in the current version of the code. Is there a hacky workaround or would I need to write my own fastAPI opentrons adapter which e.g. runs in the jupyter notebook and relays commands client/sever-side? 

# Implementation details

In best case I would like to have something like this:

```py
headers = {'content-type': 'application/json',
           'opentrons-version': '*'}
path = 'http://21.3.2.5:31950/robot/pipette'
payload = {
"target": "pipette",
"mount": "right",
"task": {"action": "aspirate",
             "volume": "100",
}
.....
}
x = requests.post(path, data = json.dumps(payload), headers=headers)
```

# Design

-

# Acceptance criteria

Being able to run an interactive (e.g. change protocol steps) simple, but full protocol from picking up a liquid and dispensing it somewhere from a remote machine without uploading a static protocol and start that.

mcous, 2022-10-04:
I'm happy to say that, as of the [6.1 release](https://github.com/Opentrons/opentrons/releases/tag/v6.1.0), such an HTTP API is available! We do not yet consider it truly public (here there be dragons), but it may be enough for your needs at this point.

After updating, see your OT-2's HTTP docs (available at `http://${your_robot_ip}:31950/redoc`) for more detailed response body documentation.

There are two main ways you can use this API:

- Run commands one at a time, live
- Queue up a number of commands and run them all

For both flows you:

1. Create a "Run" via `POST /runs`
2. Create a series of "Run Commands" via `POST /runs/:run_id/commands`

There are more than a few nuances (and required headers!) that are not super clear in the generated HTTP API docs, so if you're interested in trying this out and/or providing feedback, shoot me an email at mike at opentrons dot com and we can discuss!

closed 2022-10-04 by mcous

beniroquai, 2022-10-04:
This sounds interesting indeed! I'm still struggeling with the decision "who" should be in charge of the control workflow if there are multiple partners (e.g. opentrons + microscope - yup, still working on that..). Controlling the Robot from an external larger computer sometimes would be much easier. 

Are you planning to have the API more documented at any point @mcous ? 
Will there be a remote-control API for the opentrons? I remember registering for something but never got any information about it.