Opentrons/opentrons #7663: Accordion steps: 1 - robot calibration

asogluizzo, 2021-04-16:


emilywools, 2021-07-06:
_Outstanding tip cal questions_

tip length calibration is different than tip rack calibration. Tip length calibration is the z distance between the top of the tip and the bottom of the pipette nozzle. the value is meant to establish the value that is required for the tip to adhere to the nozzle. this is a value associated with Robot Calibration.

The measurement displayed for a tip length calibration is the z value. 

Tip rack calibration is part of the labware setup experience and is not part of robot calibration.

emilywools, 2021-07-06:
**_Outstanding deck calibration questions_**

_When if at all does a calibration "expire"? (i dont think it does, but want to confirm)_
It does not expire. If a user initiates a factory reset it will clear the deck cal and the user will have to redo it before they can proceed to upload protocols.

_What happens if deck is not calibrated? does this launch a deck calibration workflow, and if so, what does that look like?_
if the deck is not calibrated we have gating scenarios that prevent the user from proceeding. [See this figma link](https://www.figma.com/file/nbDHskbW5zaXwtrLvEIKVj/calibration-overhaul?node-id=84%3A770)
_How do we know when to display a check or an empty circle?_
empty circle:

1. there is not a store calibration value for the deck, a given pipette or a required tip length
2. the correct pipette is not attached

green check:

1. there is a stored calibration value for the item

**_Outstanding required pipettes questions_**

_Does left mount / right mount always display, and if no pipette display "none"? Or do we only display Left/Right mount if there is a pipette?_
please hold, trying to get this info for you

_If a Left or Right pipette has not been calibrated, do we kick off that flow here?_
Yes

_How do we know whether to display a check or an empty circle?_
if the correct pipette is attached (and in the correct mount), and the app has stored values, it will display a green check mark.

@b-cooper i am not sure how we handle robot calibration check on the protocol upload page today, do you happen to know?

b-cooper, 2021-07-07:
@emilywools, currently we don't expose the Robot Calibration Check anywhere in the protocol upload page. It is only accessible from the Calibration Card on the robot page. 

One part of the existing Protocol Info page that isn't yet captured in the new designs is the different callouts for missing pipettes or inexact matches (e.g. p300 Single GEN1 specified in protocol, p300 GEN2 attached).  

A missing pipette currently has a CTA to the attach pipette flow like this:
[image] 

An inexact match has an info link to an article explaining things.  link is [here](https://support.opentrons.com/en/articles/3450143-gen2-pipette-compatibility) for reference
[image]

closed 2021-09-22 by b-cooper