Opentrons/opentrons #7673: bug:

Matt-Zwimpfer, 2021-04-16:
# Overview

After pausing a protocol, the RESUME button switches to the RESET RUN Button without clicking the CANCEL button. Thus, the user can no longer resume the protocol run and will either have to reset or cancel it. 

# Steps to reproduce
Still figuring out how to reproduce this, though it seems to stem from the other [infrequent systematic pause issues](https://github.com/Opentrons/opentrons/issues/7167). 

# Current behavior
In some runs, when a user wants to manually pause by clicking the PAUSE button on the App or there is one programmed in the protocol, the pause will happen, but when the user goes to resume the protocol, the button where it should be has switched from RESUME to RESET RUN without the user cancelling the protocol run.

# Expected behavior
When a pause occurs, the RESUME button should persist and not switch to RESET RUN unless the CANCEL button is selected.

Matt-Zwimpfer, 2021-04-16:
Possible troubleshooting step: 
When you are running a protocol and you swap from the "Run" tab of the app to the "Robots" tab, it can change the "Pause" button to a "Reset" button.

Matt-Zwimpfer, 2021-04-26:
Logs from OT-2 that experienced this problem:
 
[api (45).log](https://github.com/Opentrons/opentrons/files/6378713/api.45.log)
[serial 04222021.txt](https://github.com/Opentrons/opentrons/files/6378714/serial.04222021.txt)

nonymoosee, 2021-07-19:
Any updates on this? Issues is from way way before april too.

pkMyt1, 2021-12-07:
I second the any updates on this.  I just lost half a PCR plate of ddPCR reactions because the Pause went to Reset that required me to do a forced reload of the app to clear.  I paused the run, opened the door for about 30 seconds while I was replacing a tube.  I played with it some and it seems to do this more often if the pause lasts +30 seconds and the door is opened.

nonymoosee, 2021-12-07:
Hey Dennis,
Not the worlds best solution but we basically jammed a toothpick into the
door sensor and broke it off. Easy (but silly) solution.

closed 2025-12-15 by sfoster1