Opentrons/opentrons #7698: bug: Confusion (Error?) about Z-coordinates and limits when moving pipets

beniroquai, 2021-04-22:
# Overview

We try to pipette liquids into a dish on a relatively high plateau (microscope) inside the opentrons. To not destroy the optics, we need to move the robot as high as possible and move the pipette to the objective, put it down and pipette, continue..

What happens: The first few times after `protocol.home()`, I can use deck coordinates of around 190 (which would be really nice if we can use it). After a few pipetting steps, this is not possible any more and the robot complains about hitting the limit. 
Is it an error on my side or is there some confusion inside the software? Perhaps it has to do something with this "diagonal" motion which should be fixed in 4.3? (by the way, when will this be released? 

Distance between deck and dish is around 155mm. The setup looks like this:
[image]

# Steps to reproduce

```py
from opentrons.types import Location, Point

i_primary = 0
i_secondary = 1
i_bsa = 2
i_pbs = 3
i_trash = 4
V_aspirate = 200

offset_z_reagents = -15
# define locations 
minimum_z_height = 170 # minimum height for arcs driven by the robot
position_sample_light = (150,150,190)
position_sample =  (110,120,170)
position_primary = list(plate_reagents.well(i_primary).center()._point)
position_secondary = list(plate_reagents.well(i_secondary).center()._point)
position_bsa = list(plate_reagents.well(i_bsa).center()._point)
position_pbs = list(plate_reagents.well(i_pbs).center()._point)
position_trash = list(plate_reagents.well(i_pbs).center()._point)

def move_to_coord(position_xyz, offset = (0,0,0), minimum_z_height=190):
    position_final = (position_xyz[0]+offset[0],
                        position_xyz[1]+offset[1],
                        position_xyz[2]+offset[2])
    print("Moving to: "+str(position_final))                       
    pipette.move_to(Location(Point(*position_final), None),minimum_z_height=minimum_z_height)

#ipette.home()
#
#i_pipette = 0
#i_pipette = pickup_fresh_pipette_tip(i_pipette, pipette, tiprack)

# pkcup primary
move_to_coord(position_primary, offset=(0,0,offset_z_reagents), minimum_z_height=None)
pipette.dispense()
pipette.aspirate(V_aspirate)

# add primary to sample
move_to_coord(position_sample, offset=(0,0,0), minimum_z_height=None)
pipette.dispense()
```
Error message:

```py
Moving to: (17.48, 252.67000000000002, -3.4299999999999997)
Out of bounds move: A=(191.65 motor controller, 191.65 deck) too high for limit 170.15
Moving to: (110, 120, 170)
alarm/error outside hard halt: ALARM: Hard limit +A
alarm/error: command=M907 A0.8 B0.05 C0.3 X0.3 Y0.3 Z0.1 G4P0.005 G0A191.65, resp=ALARM: Hard limit +A
Move failed
Traceback (most recent call last):
  File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1037, in _send_command
  File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1071, in _send_command_unsynchronized
  File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1110, in _handle_return
opentrons.drivers.smoothie_drivers.driver_3_0.SmoothieError: SmoothieError: None returned ALARM: Hard limit +A

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 1093, in _move
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/controller.py", line 117, in move
  File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1547, in move
  File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1054, in _send_command
opentrons.drivers.smoothie_drivers.driver_3_0.SmoothieError: SmoothieError: M907 A0.8 B0.05 C0.3 X0.3 Y0.3 Z0.1 G4P0.005 G0A191.65 returned ALARM: Hard limit +A
---------------------------------------------------------------------------
SmoothieError                             Traceback (most recent call last)
/usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.pyc in _send_command(self, command, timeout, suppress_error_msg, ack_timeout, suppress_home_after_error)

/usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.pyc in _send_command_unsynchronized(self, command, ack_timeout, execute_timeout)

/usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.pyc in _handle_return(self, ret_code)

SmoothieError: SmoothieError: None returned ALARM: Hard limit +A

During handling of the above exception, another exception occurred:

SmoothieError                             Traceback (most recent call last)
<ipython-input-88-ef09a6f8672c> in <module>()
     40 
     41 # add primary to sample
---> 42 move_to_coord(position_sample, offset=(0,0,0), minimum_z_height=None)
     43 pipette.dispense()

<ipython-input-88-ef09a6f8672c> in move_to_coord(position_xyz, offset, minimum_z_height)
     25                         position_xyz[2]+offset[2])
     26     print("Moving to: "+str(position_final))
---> 27     pipette.move_to(Location(Point(*position_final), None),minimum_z_height=minimum_z_height)
     28 
     29 

/usr/lib/python3.7/site-packages/opentrons/protocols/api_support/util.pyc in _check_version_wrapper(*args, **kwargs)

/usr/lib/python3.7/site-packages/opentrons/protocol_api/instrument_context.pyc in move_to(self, location, force_direct, minimum_z_height, speed)

/usr/lib/python3.7/site-packages/opentrons/protocols/implementations/instrument_context.pyc in move_to(self, location, force_direct, minimum_z_height, speed)

/usr/lib/python3.7/site-packages/opentrons/hardware_control/adapters.pyc in call_coroutine_sync(loop, to_call, *args, **kwargs)

/usr/lib/python3.7/concurrent/futures/_base.pyc in result(self, timeout)

/usr/lib/python3.7/concurrent/futures/_base.pyc in __get_result(self)

/usr/lib/python3.7/site-packages/opentrons/hardware_control/api.pyc in move_to(self, mount, abs_position, speed, critical_point, max_speeds)

/usr/lib/python3.7/site-packages/opentrons/hardware_control/api.pyc in _move(self, target_position, speed, home_flagged_axes, max_speeds, acquire_lock, secondary_z, check_bounds)

/usr/lib/python3.7/site-packages/opentrons/hardware_control/controller.pyc in move(self, target_position, home_flagged_axes, speed, axis_max_speeds)

/usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.pyc in move(self, target, home_flagged_axes, speed)

/usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.pyc in _send_command(self, command, timeout, suppress_error_msg, ack_timeout, suppress_home_after_error)

SmoothieError: SmoothieError: M907 A0.8 B0.05 C0.3 X0.3 Y0.3 Z0.1 G4P0.005 G0A191.65 returned ALARM: Hard limit +A
```

# Current behavior

 (see above)

# Expected behavior

Move the z-position to the highest possible position (e.g. the one after pipette.home())

SyntaxColoring, 2021-04-22:
I don't fully understand what's going on, but here's a partial breakdown of the error messages:

> ALARM: Hard limit +A

This is the actual error that's halting the run. The limit switch for the right pipette mount is unexpectedly being triggered—the mount is going too high.

> Out of bounds move: A=(191.65 motor controller, 191.65 deck) too high for limit 170.15

This is a warning, not an error, and my understanding is that it's usual to see false-positives of these messages.

*However,* it's suspicious to me that this says `191.65` instead of the `190` that you wrote in your protocol.

Strange.

mcous, 2021-04-22:
@beniroquai if you see this error message: `ALARM: Hard limit +A`, it means the robot has physically hit its limit detection switch. This is a particularly unfriendly error message, but for reference: `A` in these errors refers to the right pipette's Z motor and `Z` refers to the left pipette's Z motor.

This maximum value will (a little confusingly) change depending on the pipette that's attached on a given mount due to the length of these pipettes. The exact value will change depending on calibration value, but you can find an approximate value for each pipette type in [`pipetteNameSpecs.json`](https://github.com/Opentrons/opentrons/blob/7ca8f76a1867ee79778552aef19768997d08fc73/shared-data/pipette/definitions/pipetteNameSpecs.json) by looking for the `smoothieConfigs.homePosition` field.

For example, for a `p300_single_gen2`, the home (max) position will be around `172.15`.

beniroquai, 2021-04-22:
Thanks for the quick response! I think this:

> This maximum value will (a little confusingly) change depending on the pipette that's attached on a given mount due to the length of these pipettes. 

is the problem I faced. So an unloaded pipette has a different offset/coordinate system as a loaded one? Is that true? 
That explains why it worked in the first run where no pipettes were attached and in the second run it suddenly gave this alarm or error. 

Is there a way to go always to the highest possible position or set the `minimum_z_height` in such a way, that I don't hit the dish no matter there is a pipette attached or not?

radmanahmadi, 2021-09-08:
I am also interested in some way to call the highest possible position. Our protocol uses some very tall labware as well, and it would be nice to set a height that isnt hard coded. 

Unless your labware is just too tall, accurately defining the dimensions of the labware avoids most crashes (though you have to be careful if the second pipette is not homed).