Opentrons/opentrons #7741: bug: Canceling an infinite-looping protocol leaves the OT-2 unconnectable

SyntaxColoring, 2021-04-29:
# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

SyntaxColoring, 2021-04-29:
There's one more ticket on infinite protocols that we should find and link this to.

SyntaxColoring, 2021-05-03:
I haven't looked at the code, but here's my technical understanding of the problem, gathered from internal chats:

* We implement protocol cancelation by having `opentrons`-controlled code check an "is canceled" flag. If the protocol has been canceled, the `opentrons`-controlled code raises a special exception.
* The exception bubbles up through the user-written protocol code to terminate it.
* Protocols normally call into `opentrons`-controlled code frequently, so this works.
* On the other hand, infinite-looping protocols (and probably other kinds of "weird" protocols) will rarely, or never, call into `opentrons`-controlled code. So the exception will never be raised.
* The robot is left in a weird unconnectable state because the ongoing thread occupies one of the RPC thread slots? (I don't really get this part.)

mcous, 2022-03-04:
Fixed in v5

closed 2022-03-04 by mcous