Opentrons/opentrons #7764: bug: SSH via WiFi times out after several hours of solid connection

Matt-Zwimpfer, 2021-05-03:
# Overview
Able to establish a connection to the OT-2 via SSH over WiFi. However, after several hours of standby, the connection will be lost.

# Steps to reproduce
1. [Set up SSH connection](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2) and [connect via SSH to the OT-2 ](https://support.opentrons.com/en/articles/3287453-connecting-to-your-ot-2-with-ssh) with WiFi.
2. Keep connection on standby for several hours
3. SSH connection is lost

# Current behavior
After several hours stand by, the ot2 will disconnect due to telnet port 22 giving a time out. The user is then unable to SSH to the OT-2 unless they connect to 48888 port in browser and open a terminal in the Jupyter Notebook, and type in `netstat –a`. Then SSH connection is able to be reestablished.

This was only succeeded with the `netstat –a `command. Other commands (`ls`, `pwd`, `ping <computer ip>`) didn't work.

# Expected behavior
The SSH connection shouldn't time out over an extended period of time.

mcous, 2021-05-03:
SSH is not known for its ability to hold solid connections over a long period of time. Various server and client configurations (or good-old-fashioned network blips) can cause an SSH connection to go down, especially on the time scale of hours. This description...

> The user is then unable to SSH to the OT-2 unless they connect to 48888 port in browser and open a terminal in the Jupyter Notebook, and type in netstat –a

...certainly makes it sound like the long-lived SSH connection is doing something like chewing through resources until something at the OS level messes up.

The OT-2 has both `tmux` and `screen` available so [a user can keep a session running](https://www.tecmint.com/keep-remote-ssh-sessions-running-after-disconnection/) even while not actively connected over SSH. I think we should recommend that users use these tools if they're going to be running anything on the robot via terminal for really any amount of time over a few minutes

mattwelch, 2021-06-03:
Bug triage determination: LHM

mcous, 2021-12-13:
Closing due to lack of identifiable issue to fix. If a user requires a long-lived session over SSH, they should follow SSH best practices and use the available `tmux` or `screen` to maintain the session on the robot so they can connect and disconnect via SSH without disrupting their session