Opentrons/opentrons #7765: after update 4.2.1 gen2 pipettes arrive about 1.5 mm lower then before the update bug:

bmyury, 2021-05-03:
Hi,

We have noticed that after the update 4.2.1 –  in OT-2s  equipped  with gen2 pipettes the tip arrives at the aspiration/dispension position (bottom of 96 well plate well) roughly 1.5-2mm lower then before the update. This makes the tip bend or break

We observe this only in robots equipped with gen2. In gen1 everything is fine. In all robots the same custom defined labware is used.

Please assist.

Thank you!

Yury

# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

Gyliann, 2021-05-05:
Hello bmyury, 
Could you reach out to the support team at support@opentrons.com so we can further investigate the issue that you are seeing. Thanks!

bmyury, 2021-05-06:
i will – otherwise it seems that default setting for well_bottom_clearance for gen2 is different then for gen1

IS-SLab, 2021-05-07:
We are having the same problem with our OT-2 after the last update. What will be the solution? I do not want to downgrade the software version.

IS-SLab, 2021-05-07:
Downgrading to 4.2.0 worked well. OT support told us they are working on it.

sfoster1, 2021-05-07:
Looks like this was an issue with improperly caching tip length calibration: https://github.com/Opentrons/opentrons/pull/7778

It will be fixed in 4.3.1 which will be released early next week, 5/10 or 5/11

Gyliann, 2021-05-07:
The support team has also created an article for anyone who runs into this issue: https://support.opentrons.com/en/articles/5214816-pipettes-moving-to-incorrect-vertical-position-after-4-2-1-update

agostinofacotti, 2021-05-11:
Just verified that the bug is resolved in v4.3.1, great work!

> 
> 
> Looks like this was an issue with improperly caching tip length calibration: #7778
> 
> It will be fixed in 4.3.1 which will be released early next week, 5/10 or 5/11