Opentrons/opentrons #7767: bug: Acceessing `tipLength` on a non-tip-rack triggers a `KeyError`

mcous, 2021-05-04:
## Overview

`tipLength` is not a required property according to the labware JSON schema, but that Protocol API treats it as always present in the `tip_length` property of `Labware`. This results in a `KeyError` if you try to use `labware.tip_length` on a labware that isn't a tip rack, and therefore does not have a `tipLength` in its definition.

## Steps to reproduce

```py
from opentrons import types

metadata = {
    "protocolName": "Tip Length Key Error",
    "apiLevel": "2.0",
}

def run(ctx):
    right = ctx.load_instrument("p300_single", types.Mount.RIGHT, [tr])
    lw = ctx.load_labware("corning_96_wellplate_360ul_flat", 2)

    tip_len = lw.tip_length
```

## Expected behavior

`labware.tip_length` should do one of the following:

- Raise some sort of `LabwareIsNotTipRackError`
    - Trying to get `tip_length` of a non-tip-rack is a programmer error, so I think this makes the most sense
- Return `0.0` for non-tip-racks
    - Least drastic change, as it would not involve changing the method's return type
- Return `None` for non-tip-racks
    - This might be a breaking change, as the return type on the method would change
    - Would also make working with tip length more frustrating for actual tip racks

SyntaxColoring, 2021-05-05:
I agree that something like `LabwareIsNotTipRackError` makes the most sense for APIv2.

For APIv3, we should consider making tip racks a distinct class from other labware. See also #7552.

mattwelch, 2021-06-03:
Bug triage determination: LLH

closed 2023-06-04 by mcous