Opentrons/opentrons #7774: bug: Moving Pipetting position has no effect

beniroquai, 2021-05-06:


beniroquai, 2021-05-17:
One more thing I wanted to add:

I have updated the snippet to:

```py
def pickup_fresh_pipette_tip(i_pipette: int, pipette, tiprack, offset_pipette_rack):

    # check if pipette is attached
    try:
        if pipette._last_tip_picked_up_from is not None:
            pipette.drop_tip()
    except:
        print("Something went wrong")
    my_fresh_tip = tiprack.wells()[i_pipette]
    adjusted_location = my_fresh_tip.center().move(types.Point(x=offset_pipette_rack[0], y=offset_pipette_rack[1], z=offset_pipette_rack[2]))
    #_pipette.pick_up_tip(_adjusted_location)
    #pipette.move_to(_adjusted_location)
    try:
        pipette.pick_up_tip(adjusted_location)
    except:
        pipette.drop_tip()
        pipette.pick_up_tip(adjusted_location)

    print("My pipette number: "+str(i_pipette)+"and type: "+str(type(i_pipette)))
    return i_pipette + 1

offset_pipette_rack = (0,-5,0) # this does not have any effect while picking up a pipette
pickup_fresh_pipette_tip(1, pipette_one, tiprack, offset_pipette_rack)
```

This snippet should correct a global offset between calibrated pipette positions and deck coordinates, where `pipette_one is a 300µl gen2 single pipette and ``tiprack a 96, 200µl tiprack in slot 7, and ``offset_pipette_rack= (0,7,0)`  a  global offset to shift the whole coordinate system for the pipettes (they won't get picked up).

The expected behaviour would be that the offset has an effect, but it doesn't. Instead, I'm using 
```py
tiprack.set_calibration(Point(*offset_pipette_rack))
```
which works well (e.g. pipette gets picked up). Is this intended behaviour?

mattwelch, 2021-06-03:
Bug triage determination: MHH

mcous, 2022-06-17:
Upon investigation, [the docs](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.InstrumentContext.pick_up_tip) _specifically_ call out that passing an adjusted location to `pick_up_tip` should work:

> If the position to move to in the well needs to be specified, for instance to tell the robot to run its pick up tip routine starting closer to or farther from the top of the tip, location can be a types.Location; for instance, you can call instr.pick_up_tip(tiprack.wells()[0].top()).

Given the docs say it should work, **I am categorizing this as a bug** rather than a feature request and fixing it. As an additional data point, we allow the pick up point to be adjusted when using HTTP commands and JSONv6 protocols, so this fix would _also_ bring PAPIv2 into alignment with those systems

@beniroquai to answer your question about `set_calibraration`, it sets the offset only in the scope of the protocol run itself. That being said, I do not recommend its use. It was not documented because it was not intended for public consumption, and has since been explicitly deprecated.

Since this ticket was filed, we released a newer labware positioning check system, which hopefully obviates the need for any manual fine-tuning in the protocol itself. If you are running your protocols in Jupyter or from the command line, we've also added the [set_offset](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.labware.Labware.set_offset) method that is an officially blessed method to do what `set_calibration` was doing.

TamarZanzouri, 2022-07-14:
Tested on a robot. Works as expected. when using the supplied protocol the pipette now collides the door.
I have attached a link to the slack thread talking about this issue. It is a known issue that will be addressed in the future. 

https://opentrons.slack.com/archives/G01J0KL5XAA/p1657821663686059

```
from opentrons import protocol_api, types

metadata = {
    "apiLevel": "2.12",
}

def run(protocol: protocol_api.ProtocolContext) -> None:
    tip_rack = ctx.load_labware("opentrons_96_tiprack_300ul", 5)
    pipette = ctx.load_instrument("p300_single", types.Mount.RIGHT, [tip_rack])
    well = tip_rack.wells()[0]
    
    # offset the well to something obvious
    # to observe that the offset is not respected
    well_offset = types.Point(x=-100, y=-100 z=0)

    pipette.pick_up_tip(well.top().move(well_offset))
    right.drop_tip()
```

PR #11038 (2022-07-22): fix(papi-v2): allow pick_up_tip to pick up from exact location 
# Overview

closes #7774.
PAPIv2 documentation mentions picking up tip from a specified location.
The implementation details always moved to top.
If type is `types.Location` and `is_well()` we use specified location.

# Changelog

- Added decoy test file to test instrument_context
- Changed imports in instrument_context to use global imports 
- changed logic when pick_up_tip is called to use location when using types.Location and when the type is Well

# Review requests

1. is it ok I changed to global imports? 
3. Do we write tes [... cut]
closed 2022-07-22 by TamarZanzouri