Opentrons/opentrons #7775: feat: Allow OT-2 to continue on with protocol after temperature is set

ethanfjones, 2021-05-06:
Customers should be able to set a temperature on the temperature module and continue on with their protocol. Currently, after setting a temperature on the module the protocol pauses and waits for the temperature to be reached.

b-cooper, 2021-05-07:
This functionality should already exist for the temperature module from within the python protocol API.

Does this cover your desired use case?

the functions to use are:

- the first will set the module's target temperature and allow the protocol to continue while the module ramps up to reach that temperature.
```
    temperature_module_instance.start_set_temperature(target_degrees_celsius)
```

- the second function can be called later on to insure that the target temperature has been reached before proceeding to the following step (note: other steps can occur between the call to `start_set_temperature` and `await_temperature`)
```
    temperature_module_instance.await_temperature(degrees_celsius_to_await)
```

NOTE: this second `await_temperature` can be misused to accidentally cause an infinite hold. If the temperature passed as the one parameter is not  reached by the module, the protocol will hold indefinitely. Therefore it is very important to call `start_set_temperature` first.

ethanfjones, 2021-05-07:
Yeah, start_set_temperature would work @b-cooper . This is more of a feature request for the API docs now.

SyntaxColoring, 2021-05-19:
It looks like `start_set_temperature()` is deliberately excluded from the docs:

https://github.com/Opentrons/opentrons/blame/e446e26fcf4dcf0267cf0d221f137cc16a0265c4/api/docs/v2/new_protocol_api.rst#L33

> This will not be published to python users, it is intended to be used only by PD for the time being.

I can't figure out what mechanism is preventing `await_temperature` from showing up, though.

SyntaxColoring, 2021-09-02:
If/when we decide to officially support this, we should double-check the name. It would be easy-ish to change now, but harder to change later.

My vote is for `wait_for_temperature()`.

* It avoids conflicting with Python's special `await` keyword, which would erroneously imply `async`-ness.
* It is more likely to match UX copy, like in Protocol Designer. UX copy is more likely to say "wait for" than it is to say "await."
* The pattern `wait_for_*` is extensible to pretty much anything.

closed 2025-12-15 by sfoster1