Opentrons/opentrons #7808: Protocol Engine: Basic protocol session creation and execution

SyntaxColoring, 2021-05-13:
# Overview

From our [Protocol Engine experience milestones](https://opentrons.atlassian.net/wiki/spaces/RPDO/pages/523075585/Protocol+Engine+-+Experience+Milestones):

> As a user, I would like to start a session with a basic protocol and run that protocol.

# Background

As specified in another ticket, an endpoint like `POST /protocol_upload` (not necessarily that exact name) will let you upload a protocol file, and will give you back some kind of handle to that protocol.

# Acceptance criteria for this ticket

* [ ] You can create a *protocol session* on the robot by hitting an HTTP endpoint.
  * [ ] Creating a protocol session requires a handle to a protocol, from POST `/protocol_upload` (or whatever).
* [ ] Creating a protocol session otherwise has no runtime effect. It doesn't make the robot move.
  * Unclear: should the robot's calibration state be captured at this session creation, or should only when we actually run the protocol?
* [ ] A separate action on the protocol session resource (e.g. `POST /session/foo/start`) makes the robot start moving.
* [ ] A protocol can be started exactly once. Starting the protocol after it has already been started is an error.
	* This includes if the protocol has run to completion. You can't restart a protocol session that's finished; protocol sessions are one-shot.

# Out of scope for this ticket

* Python protocols.
* Inspecting current protocol state (the run log).
* Inspecting protocol requirements (pipettes and labware).
* Pausing, canceling, resuming.
* Protocol features deemed "advanced" for our purposes, like 8-Channel pipettes, modules, and custom labware. (Unless any of these happen to be easy.)
* Letting an HTTP client manage protocol sessions other than creating them (listing, deleting, and updating) (unless it happens to be easy).

amitlissack, 2021-05-18:
Important to keep in mind that this functionality already exists in the robot-server.

mcous, 2021-06-16:
Closing in favor of #7946, which is the last known item of work in this flow

closed 2021-06-16 by mcous