Opentrons/opentrons #7816: Protocol engine: Basic JSON protocol upload over HTTP

sanni-t, 2021-05-17:
# Overview

As an OT-2 user, I want to use an HTTP endpoint to send a single basic JSON protocol file to the robot.
Related to: #7808 

# Acceptance criteria

- [ ] a `/protocol_upload` http POST endpoint that is able to fetch the protocol file as form data
- [ ] the robot-server, upon receiving the request with the file, sends the file to Protocol runner which translates, extracts metadata and returns appropriate response* data to robot-server. The protocol runner does all things necessary to prepare the protocol in order to be run using P.E (covered in another ticket.)
- [ ] *Appropriate response to robot-server should include:
- Metadata
- Labware
- Handles to get:
-- Labware definitions
-- Protocol steps (Run log)
-- Start protocol (or protocol session)

# Out of scope for this ticket
- Features related to protocol simulation. Since this ticket relates to JSON protocols only, we should be able to implement the protocol upload w/o simulation.

amitlissack, 2021-05-18:
Important to keep in mind that this functionality already exists in the robot-server.

mcous, 2021-06-02:
This ticket was implemented by standing up a separate, feature-flagged router to avoid collision with any of the existing (working) HTTP protocol upload beta stuff.

All stuff about "metadata" in the original ticket was punted on in favor of a separate ticket about protocol simulation

PR #7856 (2021-06-02): refactor(robot-server): add experimental protocol router
## Overview

This PR is 2 of 4 working towards #7808 and #7816. Closes #7816.

1. ~chore(api,robot-server): update testing dev deps #7855~ Merged
2. **refactor(robot-server): add experimental protocol router #7856**
3. refactor(api): add create_file_runner factory #7857
4. refactor(robot-server): start wiring /sessions into runner and engine #7858

This PR adds an experimental `/protocols` router when the `enableProtocolEngine` feature flag is on. It's goal is to tackle the following checkbox from #7816:

- [ ] a /protocol_upload http PO [... cut]
closed 2021-06-02 by mcous