Opentrons/opentrons #7818: bug : The robot has to be homed manually often , protocols.home() for OT2 doesn't do the task well

SnehaMohanty, 2021-05-18:
# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

beniroquai, 2021-05-18:
Hey to clarify some pieces: The Robot performs perfectly fine, but we had a question about the homing position. 

The situation looks as follows:

[image]

and in case there is an accident (e.g., the pipette runs into an unexpended object), we need to call `pipette.home()`. The robot will first go to home-x, then home-y. Is there any command to switch these commands or tell the robot "don't go there, there is something in your way"? If something like this happens, we have to manually move the robot around the microscope inside the opentrons - this is of course not very clever if you want to have an automated protocol. 

Maybe this helps a bit to get the situation right :-)

SyntaxColoring, 2021-05-19:
Hmm.

Unfortunately, I think the order in which the OT-2 homes its axes is fundamentally tied to the robot hardware. The +Y homing switch will only be physically contacted if the pipettes are moved all the way +X before moving +Y. So X-homing needs to come before Y-homing.

(You can see this yourself if you turn the robot off and drag the pipettes by hand to the back-right corner. You'll see that if you go +Y first, the gantry will move farther back than the +Y homing switch, and get caught on the side of that switch when you move the gantry +X.)

SyntaxColoring, 2021-05-19:
From your picture, it looks like even if the pipettes are vertically homed (+Z), they'll run into that orange wall to their right. Is that a correct understanding of the problem?

beniroquai, 2021-05-19:
Hey, thanks for your comment. Indeed, the space is very tight. If the pipettes are homed and one "plans" the motion correctly, nothing happens. We tried hard to make the setup smaller, but this is the absolute minimum height we need to have in order to get the microscope working. Is there a function to override a possible homing event, e.g. by saying move -100mm relative in X and 100mm in Y from where you are right now no matter if the internal state says "please home first"? This way we could somehow make sure the robot is not hitting the microscope ;-)

SyntaxColoring, 2021-05-19:
Have you already tried calling `move_to()` with a set of known-safe deck coordinates? Or would that not work because you need to recover from a hard limit error or something, and the OT-2 is refusing to move anywhere until it's homed?

beniroquai, 2021-05-19:
It would be the latter case when the robot hit the hard limit (e.g.
microscope) and needs to do homing. I guess move_to() doesn't work then,
right?

mattwelch, 2021-06-03:
Bug triage determination: HHH.

SyntaxColoring, 2021-06-08:
> It would be the latter case when the robot hit the hard limit (e.g.
microscope) and needs to do homing. I guess move_to() doesn't work then,
right?

Not necessarily, actually. A hard limit is when the robot unexpectedly bumps into one of its own limit switches. If it "only" bumps into a microscope, the robot doesn't actually have any way of knowing that. As far as it can tell, the move completed successfully. So `move_to()` might work.

(For future hardware revisions, we're looking into ways for the OT-2 to detect when it's bumped into things on the deck. But until we do that, you can take advantage of our current hardware's stupidity. 😄 )

SyntaxColoring, 2021-06-08:
I've adjusted this issue's title and labels to reflect my understanding of the discussion above. Let me know if I've misunderstood, though!

I'm closing this issue because after discussing it internally, we don't think it's feasible for us to support changing the homing behavior. A workaround for some scenarios is given above, though.

Let me know if I can clarify anything. 🙂 

<details>
<summary>Technical details</summary>

[The firmware of our motor controller board](https://github.com/Opentrons/SmoothiewareOT) requires a home after a hard limit error. In theory, it doesn't *have* to, but since this assumption is inherited from [the upstream Smoothieware project](https://github.com/smoothieware/smoothieware), it's nontrivial to change.

Our own software has a deeply-engrained assumption that, whenever the robot needs to home, it will home the x-axis before the y-axis. As mentioned above, this is because the physical construction of the robot requires that, in the general case. Theoretically, we could make exceptions to allow for customization in specific cases like the one brought up in this issue. But we don't feel prepared to support that internal complexity.
</details>

closed 2021-06-08 by SyntaxColoring