Opentrons/opentrons #7825: feat: Allow users to select which mount side and multichannel pipette type will be used in Custom Labware test protocols

Zanecrc1, 2021-05-20:
# Overview

Currently the Custom Labware Creator only allows the use of single channel pipettes on the right mount to be used in custom labware definition test protocols. This should be expanded to include the ability to specify the left or right mount, as well as include multichannels in the pipette type drop down menu.

SyntaxColoring, 2021-05-20:
I can't find an authoritative source right now, but I remember uncertainty about whether we *could* use 8-Channel pipettes in this test protocol, I think because we thought their positioning would be less precise?

Not sure about the mount selection, though.

Zanecrc1, 2021-05-20:
True, that makes sense. Maybe this should be split into two feature requests and if the multichannel addition is not recommended we can do away with that ticket. @SyntaxColoring 

Although the purpose of this test protocol is to check "generally" that the definition is correct rather than exactly, I'm not sure how heavily adding the multichannel will stray from the intended purpose of this generated script or if those small positional inaccuracies are digestible.

closed 2025-12-15 by sfoster1