Opentrons/opentrons #7826: feat: Instrument context switching

mikolaj-w-provectusalgae, 2021-05-20:
# Overview

We are trying to speed up our processes and have found one robot behaviour that seems unnecessary at first glance.
Whenever we are changing the context of the pipette, the previous that was in use is homing to the mount (makes few small movements at the top of the mount). It takes additional time. Unfortunately, the order of activities cannot be changed to avoid context swap. Is it possible to get rid of this action?

We have a 96-well plate. We use a 20 ml pipette to transfer the sample from each well to the microscope. Before taking the sample, we must mix the liquid in the well. We use the second 1000 ml pipette for this (20 ml does not mix well). 

# Implementation details

```
pip_1000.mix(2, 260, water, rate=2.0)
# before pip_20 is used, pip_1000 goes up and makes a few moves up and down.
pip_20.aspirate(volume, water)
```

SyntaxColoring, 2021-06-03:
I don't think we have a workaround for this currently, unfortunately.

I agree with this feature request. The old pipette does probably need to retract all the way up, but I don't think we need to make it actually *home.* (I think we should make it skip those few small movements at the top.)

mikolaj-w-provectusalgae, 2021-06-05:
Thank you for your response. The elimination of "Homing" is a big improvement for us. I look forward to the changes in the next releases. Cheers

closed 2022-07-07 by laviera