Opentrons/opentrons #7831: feat: tail -f api.log

Akuli, 2021-05-24:
This came up in #4772 (cc @SyntaxColoring )

# Overview

I want to watch how `api.log` changes as I run commands on robot. I can click "Download logs" button and then look at time stamps, but it would be nicer to see things getting logged right away.

# Implementation details

I have ssh connection to the robot and I found `/data/api.log` (or maybe `/data/logs/api.log`? don't remember), but it's much shorter than the file produced by "Download all" button.

# Design

A couple options:
- Just document the correct `tail -f /some/magic/path/api.log` command and mention ssh
- Some kind of thingy in the app that you can click to see logs in real time

# Acceptance criteria

- [ ] Documentation

SyntaxColoring, 2021-05-24:
Try this when  you're SSH'd in:

```shell
journalctl --follow
```

This is equivalent to `tail -f`, except it shows logs for the whole system, not just Opentrons software. ([`journalctl`](https://man7.org/linux/man-pages/man1/journalctl.1.html) is the interface to [journald](https://www.freedesktop.org/software/systemd/man/systemd-journald.service.html), which collects all our system logs.) Try other options like `--unit=opentrons-robot-server` for filtering.

SyntaxColoring, 2021-05-24:
I agree that this would be nice to document.

I see it as part of the "OT-2 platform" or "OT-2 operating system." We would document this in the same place we document stuff like:

* Where in the filesystem hierarchy it's safe to store your own files
* How to install your own binary packages
* How to run your own service in the background

SyntaxColoring, 2024-02-21:
Closing: I don't think enough demand has materialized for a real, official, stable, documented way to tail the logs.

If you're willing to use an unofficial, undocumented, possibly unstable way, and you can [SSH into your robot](https://support.opentrons.com/s/article/Setting-up-SSH-access-to-your-OT-2), the [journalctl command](https://github.com/Opentrons/opentrons/issues/7831#issuecomment-847251249) that I listed above will still work.

closed 2024-02-21 by SyntaxColoring