Opentrons/opentrons #7878: Pipette offset problem

agostinofacotti, 2021-06-05:
Just recalibrated deck, pipette offset and tip length.
See the image attached taken using the protocol below: the 8-channel pipette has still offset...

[image]
[image]

[TestTwoPipettesAndTipracks_same_tip.zip](https://github.com/Opentrons/opentrons/files/6461667/TestTwoPipettesAndTipracks_same_tip.zip)

_Originally posted by @agostinofacotti in https://github.com/Opentrons/opentrons/issues/7793#issuecomment-839088313_

agostinofacotti, 2021-06-05:
Here a manual way to adjust the offset: inverting by hand the sign of the right pipette x-offset
https://github.com/Opentrons/opentrons/issues/7793#issuecomment-851498862

mcous, 2021-08-12:
Still working on parsing through all the data, but I tried to reproduce this last week with a full factory reset and calibration from the ground up, and I was unable to come to a conclusive response. There was a _slight_ difference between the single and multi going to the same tip rack, but not enough of a difference to interfere with tip pickup, nor was it outside our expected margins of tip variance + operator eyeball error

agostinofacotti, 2021-08-12:
Hi @mcous, thanks for your response 👍 
I think the easiest way to see some difference is when you use two P300 multi pipettes on left and right side... In that way you can see 8 tips in a row instead of only one, and it is easy to find the "median" position.
=> see the test protocol attached.

Of course, as you said, tips have dimensional variance; for that when I do the pipette calibration I pay very much attention to use the same tip orientated in the same way (writing a line on the tip, see attached image): I think that calibrating without this is less useful since you're including tip variance in pipette calibration... 

[Paired-pipette-offset-verify.zip](https://github.com/Opentrons/opentrons/files/6977431/Paired-pipette-offset-verify.zip)
[image]