Opentrons/opentrons #7913: bug: Opening door while robot paused crashes robot

JohnGoertz, 2021-06-09:
# Overview

I have a pause written into my (python) protocols where I need to open the robot door. Sometimes (but not always), this causes the robot to crash. The "run" view in the app resets from "Restart Run" and "Cancel Run" to only "Reset Run", then the robot becomes unresponsive (toggling the Lights button does nothing).

# Steps to reproduce
Write python protocol with a `protocol.pause` (this is action ~1600 in my protocol). Open the door, then close the door. The crash happens at one of those exact moments, usually when opening the door.

# Current behavior
Crashing.

# Expected behavior
Not crashing.

JohnGoertz, 2021-06-11:
Hi @SyntaxColoring, yeah the robot is is on 4.3.1 and the Robot Door Safety Switch is off.

SyntaxColoring, 2021-06-11:
Could you share your full protocol file, as well as any custom labware definitions it uses, so we can try to reproduce and narrow it down?

Preferably as a .zip file posted in this GitHub issue, but if it's confidential, you can also email it to me at max@opentrons.com.

JohnGoertz, 2021-06-12:
[Protocol for bug report.zip](https://github.com/Opentrons/opentrons/files/6641986/Protocol.for.bug.report.zip)

Here you go!

mcous, 2021-06-16:
We tested this out on our just released version 4.4.0 software, and were unable to reproduce the issue. Given that this release included fixes targeting the pause system, I'm optimistic that we may have fixed the root issue. @JohnGoertz are you able to give try the latest software and confirm whether or not you are still seeing this behavior?

JohnGoertz, 2021-06-18:
Yeah, I saw that, hopefully! Unfortunately, it is an intermittent issue, I
can't see a pattern in when it happens and when it doesn't, which I know
doesn't help things from your end. Thanks for trying. I'll update the
software and fingers crossed!

silvtal, 2021-09-16:
This just happened to me at server version 4.5.0, so not fixed.

SyntaxColoring, 2022-07-07:
Unable to reproduce on v6.0.0-beta.1 with this protocol:

```python
metadata = {"apiLevel": "2.12"}

def run(protocol):
    pipette = protocol.load_instrument("p300_multi_gen2", mount="left")
    lw1 = protocol.load_labware("opentrons_96_tiprack_300ul", 1)
    lw2 = protocol.load_labware("opentrons_96_tiprack_300ul", 3)
    for i in range(10):
        protocol.pause()
        for i in range(2):
            pipette.move_to(lw1["A1"].top())
            pipette.move_to(lw2["A12"].top())
```

And the relevant code has changed a lot since this ticket was written, so we're pretty sure this problem doesn't exist anymore.

closed 2022-07-07 by SyntaxColoring