Opentrons/opentrons #7923: Protocol engine: Update the behavior of JSON & Python protocol pause

sanni-t, 2021-06-10:
# Overview

Follow-up work for #7921 & #7922.
Update the behavior of JSON & Python protocol pause to match the current (apiv2) implementation or whatever decision is made post UX spike for pause/resume/cancel.

mcous, 2022-06-17:
Ticket under investigation as a potential 6.1.0 release requirement

mcous, 2022-06-22:
Discussed this with @SyntaxColoring and @sanni-t, going to move forward with this as a 6.1.0 requirement based on the goal of achieving as much parity as possible between JSONv6 execution and JSONv5 execution unless we have a strong reason (e.g. a bug) to avoid parity. We could not identify a strong reason to avoid parity for the 6.1.0 release, so we're going to move forward with this ticket accordingly.

We do not anticipate any architectural problems with implementing this behavior as spec'd. If we _do_ hit any show stoppers, the current behavior of command-level-pauses is ok, but would require additional documentation.

PR #10882 (2022-06-30): fix(engine): pause hardware API when engine is paused
## Overview

This PR adds calls to `hardware_api.pause` and `hardware_api.resume` when `engine.pause` and `engine.resume` are called, respectively.

Closes #7923

## Changelog

- fix(engine): pause hardware API when engine is paused

## Review requests

This PR needs to be tested on actual hardware rather than the simluator or emulator.

We need to check these acceptance criteria with both PAPIv2 protocols _and_ JSONv6 protocols, because some functionality that was scoped specifically to PAPIv2 protocols has now moved into the `ProtocolEngi [... cut]
closed 2022-06-30 by mcous