Opentrons/opentrons #7924: bug: logs do not download

nusrat813, 2021-06-10:
# Overview

4.4.0-alpha.0 logs do not download 

# Steps to reproduce

1. Click download logs on the app 
2. Hit save file for serial log 
3. Spinner spins indefenitly

SyntaxColoring, 2021-06-10:
Reproduced on my robot. Investigating now.

SyntaxColoring, 2021-06-10:
I think this was an accident of #7632.

Prior to #7632, `GET /logs` was an endpoint (possibly the *only* endpoint?) that didn't require the `Opentrons-Version` header.

https://github.com/Opentrons/opentrons/blob/13a87747a77ece0e0a25745fcda41cab947f60e1/robot-server/robot_server/service/legacy/routers/__init__.py#L40-L44

Since #7632, we universally require the `Opentrons-Version` header on *all* endpoints.

https://github.com/Opentrons/opentrons/blob/2b249d9cc24e06afd3d4e68fe85b5900fd385b18/robot-server/robot_server/app.py#L64-L68

And the Opentrons App apparently does not set `Opentrons-Version` when downloading logs. So:

```
Jun 10 16:34:12 red-dust opentrons-api[246]: Error response: 422 - LegacyError - header.opentrons-version: field required
```

We should:

* Either change robot-server to the old behavior where it doesn't require `Opentrons-Version` on this endpoint, or change the Opentrons App to set `Opentrons-Version` on its log download requests, or both.
    * Any opinions which approach we take?
* Make sure we didn't made this mistake with any other endpoints.
* Make the Opentrons App show some kind of error indicator when it gets an HTTP error downloading logs, instead of showing an infinite spinner.

mcous, 2021-06-10:
> - Either change robot-server to the old behavior where it doesn't require Opentrons-Version on this endpoint, or change the Opentrons App to set Opentrons-Version on its log download requests, or both.
>
>    - Any opinions which approach we take?

We are constrained by Electron here; the [underlying file download API](https://github.com/Opentrons/opentrons/issues/7878) does not provide a headers option (electron/electron#12883)

> Make sure we didn't made this mistake with any other endpoints.

Same story for the root RPC endpoint, fixed by #7927 

> Make the Opentrons App show some kind of error indicator when it gets an HTTP error downloading logs, instead of showing an infinite spinner.

This also appears to be a problem with either the underlying Electron file download or the [electron-dl](https://github.com/sindresorhus/electron-dl) library that we depend on. Our code, as written, fully intends to catch errors to stop the spinner. Note the `catch` followed by the `dispatch({ type: 'shell:DOWNLOAD_LOGS_DONE' })`.

https://github.com/Opentrons/opentrons/blob/8bdd2236cb09fc497f1884441605117521378478/app-shell/src/robot-logs.ts#L21-L33

closed 2021-06-14 by mcous