Opentrons/opentrons #7959: bug: Update from 4.3.1 -> 4.4.0 adds ThreadManagerException to software

Koeng101, 2021-06-18:
4.3.1 -> 4.4.0

# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

amitlissack, 2021-06-18:
Sorry that you've run into this. We are currently in a long holiday weekend, but can put time into this next week.

A couple questions:
1) Do you run into the same issue when using a simulator?
2) Sadly the the ThreadManagerException is not super helpful. The ThreadManager is simply calling this method `API.build_hardware_controller()`. For the sake of seeing the actual exception, could you call `API.build_hardware_controller()` in your code and share the exception raised?

Koeng101, 2021-06-22:
Hey @amitlissack meant to run this today, but was taking a long holiday weekend in the other direction. Will test (2) early tomorrow morning.

I do not get the same issue when running in a simulator.

d-raith, 2021-06-22:
I probably can confirm the issue from another perspective. I'm currently using jupyter notebooks to implement custom remote control, similar as how it is described in the documentation section regarding jupyter notebooks. We use a gen 2 temperature module and since the latest update we are unable to obtain a protocol context through the get_protocol_api method. It fails with the same error mentioned in the initial post but in addition I have a stack trace regarding the temperature module communication:

```
Exception in Thread Manager build
Traceback (most recent call last):
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/thread_manager.py", line 152, in _build_and_start_loop
  File "usr/lib/python3.7/asyncio/base_events.py", line 573, in run_until_complete
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 191, in build_hardware_controller
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/module_control.py", line 36, in build
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/module_control.py", line 116, in register_modules
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/module_control.py", line 63, in build_module
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/modules/utils.py", line 37, in build
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/modules/tempdeck.py", line 79, in build
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/modules/tempdeck.py", line 241, in _connect
  File "usr/lib/python3.7/site-packages/opentrons/drivers/temp_deck/driver.py", line 258, in get_device_info
  File "usr/lib/python3.7/site-packages/opentrons/drivers/temp_deck/driver.py", line 353, in _get_info
  File "usr/lib/python3.7/site-packages/opentrons/drivers/temp_deck/driver.py", line 299, in _send_command
AssertionError: not connected
```
The assertion error is triggered due to the temperature module's lock not being available / initialized. 

However, control and access of the module using the opentrons app is working. With the module being powered off, I'm able to obtain a ProtocolContext and control the robot. @amitlissack, do you have any additional modules connected to your test setup?

Info: 
Server Version: 4.4.0
Firmware Version: v1.1.0-25e5cea

Jupyter Cell:

```
import opentrons.execute
protocol = opentrons.execute.get_protocol_api('2.11')
protocol.home()
```
```
Failed to initialize character device, will not be able to control gpios (lights, button, smoothiekill, smoothie reset). Only one connection can be made to the gpios at a time. If you need to control gpios, first stop the robot server with systemctl stop opentrons-robot-server. Until you restart the server with systemctl start opentrons-robot-server, you will be unable to control the robot using the Opentrons app.
Exception in Thread Manager build
Traceback (most recent call last):
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/thread_manager.py", line 152, in _build_and_start_loop
  File "usr/lib/python3.7/asyncio/base_events.py", line 573, in run_until_complete
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 191, in build_hardware_controller
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/module_control.py", line 36, in build
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/module_control.py", line 116, in register_modules
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/module_control.py", line 63, in build_module
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/modules/utils.py", line 37, in build
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/modules/tempdeck.py", line 79, in build
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/modules/tempdeck.py", line 241, in _connect
  File "usr/lib/python3.7/site-packages/opentrons/drivers/temp_deck/driver.py", line 258, in get_device_info
  File "usr/lib/python3.7/site-packages/opentrons/drivers/temp_deck/driver.py", line 353, in _get_info
  File "usr/lib/python3.7/site-packages/opentrons/drivers/temp_deck/driver.py", line 299, in _send_command
AssertionError: not connected
---------------------------------------------------------------------------
ThreadManagerException                    Traceback (most recent call last)
<ipython-input-4-03058a0ad874> in <module>()
      1 import opentrons.execute
      2 import time
----> 3 protocol = opentrons.execute.get_protocol_api('2.11')
      4 
      5 

/usr/lib/python3.7/site-packages/opentrons/execute.pyc in get_protocol_api(version, bundled_labware, bundled_data, extra_labware)

/usr/lib/python3.7/site-packages/opentrons/hardware_control/thread_manager.pyc in __init__(self, builder, *args, **kwargs)

/usr/lib/python3.7/site-packages/opentrons/hardware_control/thread_manager.pyc in managed_thread_ready_blocking(self)

ThreadManagerException: Failed to create Managed Object
```

amitlissack, 2021-06-22:
@d-raith thanks for all that info. 

By any chance is the environment variable `ENABLE_VIRTUAL_SMOOTHIE` defined in your jupyter setup?

d-raith, 2021-06-22:
I've already downgraded the system to 4.3.1 in order to continue development. The variable is not set in this configuration and  it's probably the same for 4.4.0 as long as the default behavior didn't change with the update as we do not make use of it in our code.

amitlissack, 2021-06-22:
@d-raith  `ENABLE_VIRTUAL_SMOOTHIE` was a long shot but an easy explanation. I do not have a tempdeck to test with. I'll continue digging. 

Can you verify that you have only have one tempdeck connected?

d-raith, 2021-06-22:
@amitlissack Yes, there is only one temp deck connected.

Koeng101, 2021-06-22:
I have a temp deck connected as well. I can disconnect it and try again

sfoster1, 2021-06-22:
If you stop the opentrons robot server on the raspberry pi with `systemctl stop opentrons-robot-server` does that help things?

d-raith, 2021-06-22:
@sfoster1 I tried it, but it didn't help at least for my part. Given that @Koeng101 also uses a temp module, I'm pretty certain that this is the same or at least a closely related issue.

Russell-Tran, 2021-06-22:
Following up on behalf of @Koeng101 (same lab): 
- @amitlissack: Temp module is related to the issue. Our custom software, opentronsfastapi, is unsuccessful (fails with `ThreadManagerException`) in any instance where a Temp module is on and connected to the OT-2. That is, fails regardless of whether the particular protocol actually declares the Temp module in code (`ctx.load_module("temperature module", 4)`). Disconnect the Temp module, and opentronsfastapi for protocols which don't need a Temp module regains functionality.

amitlissack, 2021-06-22:
We have been able to reproduce the problem in our lab. We will update you all when we know more.

closed 2021-06-28 by Laura-Danielle

Koeng101, 2021-06-28:
@Laura-Danielle Great to know this is fixed! Will there be an update release soon to fix this bug? Thanks

Laura-Danielle, 2021-06-28:
Hey @Koeng101, we will be doing our next release around the end of July.

d-raith, 2021-06-29:
Glad to hear that the issue has been resolved, looking forward to the release. 
Thanks a lot for the quick response and great soft- and hardware!

PR #8009 (2021-06-28): fix(api): fix module path for jupyter notebook
# Overview

Closes #7959. We should use the absolute path for modules, otherwise jupyter notebook will not be able to find the correct port to connect to the module.

# Changelog
- Change port path from `dev/ot_module*` to `/dev/ot_module*`

# Review requests
Pls make sure this didn't break anything. Run module protocol in the app and also in jupyter notebook. Disconnect/Reconnect modules a few times.

# Risk assessment
Medium.