Opentrons/opentrons #7960: feat: manually drop tip from Opentrons app

Russell-Tran, 2021-06-18:
# Overview

As a user, I want to be able to manually instruct the OT-2 from the App to drop its tip. There are two cases where the need for this consistently pops up. 

The first is when I only want to recalibrate some of the plates for a protocol (but not all), hence the robot will grab a tip from the rack, do some calibrations, but neither return the tip nor drop it in the trash (since returning the tip to its rack only happens if you complete the entirely of protocol calibration). 

The other is when I want to abort a currently running protocol, and thus the robot homes with a tip still attached.

The current way I have to remove the tip from the robot is by physically removing the tip with my glove. This is not ideal because it results in touching the pipette, which is 1) a contamination risk and 2) a risk of messing with the calibration of the pipette.

Issue #4877 might be related, from reading its title (since I can't read the link in the body, of course).

# Implementation details

Add button to Opentrons App
Have robot drop tip

# Design

E.g., the drop tip button could be put in the App near the Home button and potentially also be grayed out unless the Robot is at Home (so that the drop happens above the trash).

# Acceptance criteria

Any means by which a user can easily command the Opentrons to drop its current tip so that one doesn't have to touch the pipette to remove it.

Russell-Tran, 2021-06-19:
A workaround is currently possible a la 
```
"""
Drop single channel tip if there's a tip that's still attached 
Ensure that deck slot 9 is empty since we will "ghost" pick up there
"""
metadata = {"apiLevel": "2.9"}

def run(ctx):
    tip_rack = ctx.load_labware("opentrons_96_filtertiprack_20ul","9")
    p20s = ctx.load_instrument("p20_single_gen2", "left", tip_racks=[tip_rack])
    ctx.home()
    p20s.pick_up_tip()
    p20s.drop_tip()
```

But this takes 55 seconds just to drop 1 tip.

SyntaxColoring, 2021-09-09:
**After thinking about this, we’re declining to add this feature (for now). Instead of having a manual button to eject the tip, we prefer for the robot to eject the tip automatically, when needed.**

Most cases of a tip erroneously remaining on the pipette are because the robot was interrupted in the middle of some process. When that happens, we think the robot should be able to clean up after itself, rather than leave itself in a weird state that the user is responsible for fixing.

> As a user, I want to be able to manually instruct the OT-2 from the App to drop its tip. There are two cases where the need for this consistently pops up.
>
> The first is when I only want to [...]

This case is surprising to me, and sounds like an oversight in our labware calibration flow. Labware calibration should never leave a tip attached, even if you exit it before visiting every labware. I’ll ticket this separately [#8326].

> The other is when I want to abort a currently running protocol, and thus the robot homes with a tip still attached.

This specific case will be fixed in an upcoming release—see #8229. Cancelling a protocol will automatically drop any attached tips in the fixed trash.

-----

This automatic cleanup strategy that we’re favoring isn’t perfect. For example, if you power off the robot while it has a tip attached, and then power it back on, you will still need to manually pull the tip off with your gloved hand. (For unfortunate technical reasons, it’s difficult for the robot to safely eject a tip that was on the pipette when the robot booted.) If cases like that turn out to be a problem, we can revisit this proposal to add a manual eject button to the app.

closed 2021-09-09 by SyntaxColoring