Opentrons/opentrons #8033: Expose errors other than Protocol Engine errors through the HTTP API

SyntaxColoring, 2021-06-30:
# Overview

Today, our experimental HTTP API correctly shows errors that cleanly fit inside a single Protocol Engine command. That individual command is marked as `failed` on the session resource.

However, this doesn't account for errors that happen in between Protocol Engine commands. For example, if a Python protocol has something like this:

```python
protocol.aspirate(123)
function_that_raises()
protocol.dispense(123)
```

We couldn't mark it as a failure on the `AspirateCommand`, because we would have already marked that command as succeeded. And we couldn't mark it as a failure on the `DispenseCommand`, because that command wouldn't have been created by the time the exception gets raised.

The HTTP API needs a way of exposing errors that happen outside the execution of a Protocol Engine command.

# Implementation details

TBD!

# Design

TBD!

# Acceptance criteria

* The client can identify where in the protocol source the error came from. e.g. a Python line number.
* The client can inspect the protocol's progress at the time of failure.

SyntaxColoring, 2021-06-30:
A simple solution would be to add top-level `status` and `errorDescription` fields on the session resource.

```yaml
status: "internalError"
errorDescription: "blah blah"
commands:
    - id: "abc"
      type: "aspirateCommand"
      status: "succeeded"
    - id: "def"
      type: "dispenseCommand"
      status: "succeeded"
    - id: "ghi"
      type: "aspirateCommand"
      status: "succeeded"
```

*However.* If we anticipate the needs of the run log in the Opentrons App and other clients:

* We know that a step in the run log will not necessarily pair 1:1 with a Protocol Engine command.
* [I think it would pretty cool](https://community.opentrons.com/t/custom-messages-in-the-log/36/5?u=opentrons_max) if Python protocol authors could customize how their code gets partitioned into steps in the run log.

Given those requirements, we'll probably want to represent the protocol as a tree of nested steps instead of a flat list, anyway. And if we're doing that, we may as well incorporate our error reporting into that structure, like this:

```yaml
steps:
    - userFacingDescription: "Transferring from A1 to B1."
      status: "succeeded"
      children:
        - userFacingDescription: "Aspirating from A1."
          status: "succeeded"
          engineCommands:
            - id: "abc"
              type: "aspirateCommand"
              status: "succeeded"
        - userFacingDescription: "Dispensing to B1."
          status: "succeeded"
          engineCommands:
            - id: "def"
              type: "aspirateCommand"
              status: "succeeded"
    - userFacingDescription: "Transferring from A2 to B2."
      status: "internalError"
      errorDescription: "blah blah"
      children:
        - userFacingDescription: "Aspirating from A2."
          status: "succeeded"
          engineCommands:
            - id: "abc"
              type: "aspirateCommand"
              status: "succeeded"

mcous, 2021-06-30:
I think I would imagine something more simple. I'm pretty wary of nesting in this manner; I think it'll make this API more difficult to both work with and maintain. I would imagine something like this:

```js
{
  data: {
    id: "session-id",
    sessionType: "protocol",
    // fed by protocol engine
    commands: [
      {
        id: "command-id-1",
        commandType: "loadLabware",
        ...
      },
      {
        id: "command-id-1",
        commandType: "loadPipette",
        ...
      }
    ],
    // fed by user
    actions: [
      {
        id: "action-id-1",
        actionType: "start",
        ...
      }
    ]
    // fed by protocol runner
    errors: [
      {
        id: "error-id-1",
        errorType: "pythonExecutionError",
        details: {
          ...
        }
      }
    ]
  }
}
```

commit 69247466 (2021-11-15): refactor(engine): store external runner errors in the engine

Closes #8033
PR #8771 (2021-11-17): refactor(engine): store external runner errors in the engine
## Overview

Building off the work in #8512, this PR stores any `ProtocolRunner`-sourced errors in the engine, allowing protocol file raises outside a command to be reported in protocol analyses and runs.

Closes #8033

## Changelog

- refactor(engine): store external runner errors in the engine

## Review requests

- [ ] Code looks good
- [ ] Tests look good
- [ ] Try to break this

### Caveats / punts

- Errors in "legacy" protocols are currently reported twice
    - We capture the error from the `after` message, but the legacy run [... cut]
commit c9c55b49 (2021-11-17): refactor(engine): store external runner errors in the engine (#8771)

Closes #8033
closed 2021-11-17 by mcous