Opentrons/opentrons #8041: UX validation and recommendations surrounding pause/cancel effects

readywater, 2021-07-01:
**As a user, I would like to understand how a pause/cancel affects the robot (instant stop versus stop when current task is complete)**
## Story
As a user, I would like to pause the robot so I can manipulate things on the deck
As a user, I would like to halt the robot if something is happening that shouldn't be happening.
As a user, I don't want to mix up pausing and halting.
[Building on top of recommendations here](https://app.zenhub.com/workspaces/opentrons-cpx-5e1f6ed55b295a3f514f2283/issues/opentrons/opentrons/531)

## Acceptance
Completed short survey sent to existing HTTP api cohort. Survey to walk through scenarios. 
$20 compensation per completed survey up to 5 respondants.

## Deliverable
Revised UX recommendations for #531 re: Pause/Play

readywater, 2021-07-22:
Survey is in flight w/ 15 participants, however we've yet to get any responses. Will be sending out reminders on the 26th.

SyntaxColoring, 2021-07-30:
Quick copy-paste of [a comment that I left on PR #8161](https://github.com/Opentrons/opentrons/pull/8161#pullrequestreview-719224967):

> I don't have a strong use case for this yet, but I've been thinking we should have three separate actions:
> 
> 1. A "pause from the outside" action. Issued by HTTP clients, and by pressing the OT-2's front button.
> 2. A resume action that undoes (1).
> 3. A resume action that undoes pauses that are built into the protocol.
> 
> Because I think an HTTP client should be able to naively undo its own pause and be confident that the robot will be left exactly how it was before, without knowing anything deep about the protocol.
> 
> For example, say you're writing a client to issue HTTP commands to the OT-2 to coordinate it with other equipment. You want to implement a "stop everything" button. You also want to implement a button that resumes from the "stop everything."
> 
> If the OT-2 was in a protocol-issued pause state when the user clicked "stop everything," you probably *wouldn't* want the OT-2 to start moving on its own when they click resume. You'd expect the OT-2 to go back to doing exactly what it was before—and what it was doing was waiting.
> 
> Currently, to work around this, you'd have to know that the robot is currently on a protocol-issued pause, and, as a special case, avoid resuming it.
> 
> Edit: And I haven't thought about how this interacts with the door opening and closing.

@readywater Let me know if you want to talk through this. It's still kind of fuzzy in my head, so I'm probably not explaining it well.

readywater, 2021-08-06:
> > A "pause from the outside" action. Issued by HTTP clients, and by pressing the OT-2's front button.
> 
> A resume action that undoes (1).
> 
> A resume action that undoes pauses that are built into the protocol.
> 
> Because I think an HTTP client should be able to naively undo its own pause and be confident that the robot will be left exactly how it was before, without knowing anything deep about the protocol.
> 
> For example, say you're writing a client to issue HTTP commands to the OT-2 to coordinate it with other equipment. You want to implement a "stop everything" button. You also want to implement a button that resumes from the "stop everything."

I think in this situation, you'd want to be able to pass some arguments to the API that could modify that behaviour, but the default pause behaviour should be consistent across interaction modes — api, button, app, door.

> 
> If the OT-2 was in a protocol-issued pause state when the user clicked "stop everything," you probably wouldn't want the OT-2 to start moving on its own when they click resume. You'd expect the OT-2 to go back to doing exactly what it was before—and what it was doing was waiting.

If you have an API issued pause and open the door, then the door interaction wouldn't affect anything.

I think though that like, if you think of being paused as being a state that the robot is in (and I know using the word state here is probably fraught), then any action to modify that state would apply. So if you pause the robot with the button but then resume it from the app, or pause the robot with an API call then press the button to unpause — that should be consistent.

> Currently, to work around this, you'd have to know that the robot is currently on a protocol-issued pause, and, as a special case, avoid resuming it.
> 
> Edit: And I haven't thought about how this interacts with the door opening and closing.
> 
>

mcous, 2021-11-22:
This ticket is awaiting survey results, if they exist. Will post and close accordingly

SyntaxColoring, 2024-02-23:
Closing as stale.

closed 2024-02-23 by SyntaxColoring