Opentrons/opentrons #8085: feat: Implement button press behavior

mattwelch, 2021-07-13:
Implement behavior defined in #531

mcous, 2021-08-09:
Ping @readywater: I was testing the button test prototype branch over in #8218, and while actually interacting with the robot, a few states shook out that I think need consideration in terms of what the LED and/or app indicates to the user. I tried to make a table below with all states, including ones listed out in the proposal:

| Robot State                      | Next user step               | Button interactable? | Current LED | Proposed LED  |
| -------------------------------- | ---------------------------- | -------------------- | ----------- | ------------- |
| Powered on                       | Load a protocol              | No                   | Solid blue  | ???           |
| Protocol loaded                  | Start the protocol           | Yes                  | Solid blue  | ???           |
| Start requested, not yet running | Wait for robot to start      | No                   | Solid blue  | ???           |
| Protocol running                 | Pause/stop the protocol      | Yes                  | Solid blue  | Solid blue    |
| Pause requested, not yet paused  | Wait for robot to pause      | No                   | Sold blue   | ???           |
| Protocol paused                  | Play/stop the protocol       | Yes                  | Solid blue  | Flashing blue |
| Waiting for stop hold            | Keep holding button          | Yes                  | Solid blue  | Flashing red  |
| Protocol stopped                 | Reset or load other protocol | No                   | Solid blue  | Solid red     |

I left two entries out of that list that are present in the most pedantic sense, but are effectively "instant":

- Resume requested, but not yet resumed
    - Resume happens right away once you hit the button
- Cancel requested, but not yet cancelled
    - Once the 2 second hold timer finishes, the robot stops moving right away

mcous, 2021-09-08:
@readywater based on what I can tell, pulsing is prohibitively difficult with the way the OT-2 hardware is wired up

readywater, 2021-09-10:
@mcous Is it possible to do a sawtooth waveform? Eg gradually brighten to 100% then turn off? Main thing is just differentiating.

mcous, 2021-09-10:
@readywater unfortunately, due to hardware constraints (i.e. no hardware PWM available), we have no easy dimming ability. Each color channel is either on or off.

I just did a quick experiment with bit-banging the GPIO pin from the Python software to get some pulsing. We could probably get something serviceable out of it, but it doesn't look great. My prototype relies on the timing of `time.sleep`, which is not consistent and can result in random little flashes when the light should be dim. I'm also a little wary of software PWM chewing through limited CPU resources