Opentrons/opentrons #8090: LC: don't mock ALL utils in a test

IanLondon, 2021-07-14:
# Overview

Unless there's a specific reason, it's a bad practice to mock out `utils/` because that broadly replaces all utils with mocks. This can cause false positives and negatives depending on implementation details that shouldn't be the concern of the tests

# Acceptance Criteria

Audit all `jest.mock` calls of `utils` and eg instead of mocking the entire utils directory, mock just the functions you actually care about mocking, eg `jest.mock('../../../utils/isEveryFieldHidden')`.

mcous, 2021-07-14:
Hard disagree. Mocking things out partially is a bad idea. I'm not totally familiar with the codebase, but if the problem is "`utils` is hard to mock", the root problem is not the mocking but either the tests, the dependency module, or the test subject itself. `utils` looks to contain a few entirely unrelated pieces of logic. If it can't be mocked wholesale without breaking the tests, either:

1. The tests are not written to be solely [collaboration tests](https://github.com/testdouble/contributing-tests/wiki/Collaboration-Test)
2. The test subject is not written to be solely a collaborator, and is instead mixing delegation with actual logic
3. The dependency is not scoped to do one specific thing or set of things

With anything named "utils", I'd say (3) is definitely in play. I'm not super familiar with this codebase, but I'd say (1) and (2) are likely in play, as well.

I post this talk all over the place, and I'll probably keep posting it for a while, but definitely give it a watch. It goes over my points above with much more eloquence (and actual examples) than I am capable of: https://blog.testdouble.com/talks/2018-03-06-please-dont-mock-me/

mcous, 2021-09-07:
Closing after discussion in Slack. The root of the problem isn't mocking, but the organization of unrelated functions into a `utils` bag

closed 2021-09-07 by mcous