Opentrons/opentrons #8094: bug:

danielpgp1012, 2021-07-15:
# Overview

I expect that if I provide argument blow_out_destination='source well' to a distribute command, the blow out will be done in the source well, but currently it continues to do it in the trash. API Level is 2.8.

# Steps to reproduce

```
single_300.distribute(
        vol,
        src,
        [well.top(-1) for well in all_wells],
        new_tip="once",
        mix_before=(5, multi_300.max_volume),
        disposal_volume=20,
        blow_out_location="source well"
    )
```

# Current behavior
Produces the following simulated logs:
```
Distributing 60.0 from A1 of bead-tubes on 7 to A1 of central-plate on 5
	Transferring 60.0 from A1 of bead-tubes on 7 to A1 of central-plate on 5
		Picking up tip from A1 of Opentrons 96 Tip Rack 300 �L on 8
		Mixing 5 times with a volume of 300.0 ul
			Aspirating 300.0 uL from A1 of bead-tubes on 7 at 120.0 uL/sec
			Dispensing 300.0 uL into A1 of bead-tubes on 7 at 120.0 uL/sec
			Aspirating 300.0 uL from A1 of bead-tubes on 7 at 120.0 uL/sec
			Dispensing 300.0 uL into A1 of bead-tubes on 7 at 120.0 uL/sec
			Aspirating 300.0 uL from A1 of bead-tubes on 7 at 120.0 uL/sec
			Dispensing 300.0 uL into A1 of bead-tubes on 7 at 120.0 uL/sec
			Aspirating 300.0 uL from A1 of bead-tubes on 7 at 120.0 uL/sec
			Dispensing 300.0 uL into A1 of bead-tubes on 7 at 120.0 uL/sec
			Aspirating 300.0 uL from A1 of bead-tubes on 7 at 120.0 uL/sec
			Dispensing 300.0 uL into A1 of bead-tubes on 7 at 120.0 uL/sec
		Aspirating 260.0 uL from A1 of bead-tubes on 7 at 120.0 uL/sec
		Dispensing 60.0 uL into A1 of central-plate on 5 at 120.0 uL/sec
		Dispensing 60.0 uL into B1 of central-plate on 5 at 120.0 uL/sec
		Dispensing 60.0 uL into C1 of central-plate on 5 at 120.0 uL/sec
		Dispensing 60.0 uL into D1 of central-plate on 5 at 120.0 uL/sec
		Blowing out at A1 of Opentrons Fixed Trash on 12
```

Blow out is being done at the trash even though the parameter instructed to do it at the source well.

# Expected behavior

I would expect the last line to read `Blowing out at A1 of bead-tubes on 7`

SyntaxColoring, 2021-09-21:
Hey @danielpgp1012, sorry for taking so long to get back to you on this.

I think this is because one of the parameter names in misspelled. Try `blowout_location=...` instead of `blow_out_location=...`.

(The API *should* raise errors for misspelled parameter names but because of a bug, it silently ignores them. That's issue #5106).

SyntaxColoring, 2021-09-21:
I'm marking this issue as closed now since I think that's the solution, but feel free to keep replying here if that turns out to not work.

closed 2021-09-21 by SyntaxColoring