Opentrons/opentrons #8134: feat: Upgrade version of pydantic?

danwinkler, 2021-07-19:
I'm trying to use the opentrons python package and [strawberry](https://github.com/strawberry-graphql/strawberry) in the same project, and I'm running into an issue where strawberry requires [pydantic](https://github.com/samuelcolvin/pydantic) at version 1.6.1 or higher. Opentrons pins it to 1.4. How difficult would it be to bump that version?

Thanks!

(I'm also looking into getting strawberry to play nice with an older version of pydantic, as pydantic is an optional package for strawberry.)

mcous, 2021-07-20:
Hey @danwinkler, we've been exploring this ourselves! For a period of time (>1.4, <= 1.6) we were unable to upgrade Pydantic due to lack of `TypedDict` support. Since then, I _think_ Pydantic may have added this support, and we have reduced our reliance on TypeDict.

I'll update this post if we do any reasearch, and feel free to add any of your own if you have it! For reference, corresponding ticket in the OS repository: Opentrons/buildroot#107

SyntaxColoring, 2021-07-22:
Labeling this `cpx` so it shows up ini the CPX squad's backlog, but this isn't exclusively a CPX thing; anyone should feel free to work on it, I think.

mcous, 2021-08-25:
Research update: after we get #8228 and Opentrons/buildroot#131 in, it looks like we have a pretty clear path on this upgrade. We can bundle a FastAPI and Pydantic upgrade, with the following notes:

- Newer version of FastAPI errors if the OpenAPI Spec that it generates is invalid (good!)
- Newer versions of Pydantic + FastAPI actually behave with string enums in various places (good!)
- Our generated OpenAPI Spec is **currently** invalid, and upgrading causes those errors to be surfaced (bad!)

The source of the error is that the JSONSchema output by Pydantic for `Tuple[float, float]` (for example) is not valid OAS, even though it's valid JSONSchema. We can switch these to `dataclass`s with named members, which is probably a good idea on its own, and it's not an overwhelming diff. Once that change is made, upgrading Pydantic and FastAPI appears to be unblocked

mcous, 2021-09-17:
We just barely missed the cutoff of the upcoming 4.6 release, so this upgrade will be in 4.7

PR #8345 (2021-09-17): feat(api,robot-server): upgrade fastapi and pydantic deps
## Overview

This PR upgrades Pydantic and FastAPI across all Python projects in the monorepo, and makes the necessary changes in `api` and `robot-server` to make sure everything continues to function. Closes #8134.

This PR has an accompanying Opentrons/buildroot PR: Opentrons/buildroot#137. If you test on a robot or RPi and want to approve this change, please be sure to 👍 that PR, too

## Changelog

- feat(api,robot-server): upgrade fastapi and pydantic deps

The primary changes that needed to be made revolved around the use of `typin [... cut]
commit 9cf416a4 (2021-09-17): feat(api,robot-server): upgrade fastapi and pydantic deps (#8345)

Closes #8134
closed 2021-09-17 by mcous