Opentrons/opentrons #8145: UX Spike: Accessing subsets of channels/wells

readywater, 2021-07-22:
Well access: how/why do users access a subset of channels/wells (picking up a tip on only certain channels, eg)
__Relative to 8 channel?__
## Story
As a user, I want specific and easy control over what well(s) I access at any given time. 

## Acceptance
3-5 examples from interviews tied to this question

## Deliverable
Share of research findings with interpretation (1 slide)
Use cases + user stories
Specific examples

readywater, 2021-08-23:
Notes from chat:
* Masking as an alternative (well array, pipette array)
* Transform along 0,0 to set base of mask
* Operating on the well level
* Alternative coord system?

mcous, 2022-02-07:
Closing as stale

closed 2022-02-07 by mcous