Opentrons/opentrons #8146: UX Spike: Multi-file protocol upload with HTTP API

readywater, 2021-07-22:
How should multi-file protocols and data files be accommodated as part of the http interface?

## Story
As a user, I would like to have data and dependencies like labware calibration and external to my protocol file, but easily transfered and referenced by the OT2

## Acceptance
Interviews with n number of users

## Deliverable
Share of research findings with interpretation (1 slide)

readywater, 2021-08-31:
What we should do (long term):
* Package.json like file or frontmatter which wrangles dependencies after upload and provides a clear error if dependencies are not met

What we can do now (maybe):
* Zip different files on client side (we provide code samples/example project for doing this in Node and Python)
* API consumes a tarball and unzips on robot and checks dependencies

mcous, 2021-08-31:
Still open question: if a user uploads multiple Python files, how do we decide which files is the "main" file, especially in the absence of a manifest.

Some ideas:

- In the absence of a manifest, our system will consider `__init__.py` to be the "main" file if multiple Python files are given.
    - This means we treat a set of Python files as a regular Python "package". 
- In terms of a manifest, we should heavily consider going the "standard Python" route and allow users to configure their protocol using Opentrons-specific configs in a [pyproject.toml](https://www.python.org/dev/peps/pep-0518/) file

readywater, 2021-09-10:
IMO I think the above should be moved forward on.

closed 2021-09-21 by mattwelch