Opentrons/opentrons #8147: UX Spike: Managing Simulation w/ HTTP API

readywater, 2021-07-22:
As a user, I would like my uploaded protocol to be “simulated” to provide client-side and calibration information.

A UX spike to fall out of this could be asynchronous/long running upload

## Story
As a user who has uploaded a protocol to the robot, I want a clear indication as to the status of my robot and the protocol so that I'm not left wondering if something went wrong.

## Acceptance
Consistent UX recommendations
Deliverable
Wireframes and user flow

mcous, 2021-07-27:
Sync with @emilywools so @readywater can incorporate recommendations into CPX work

readywater, 2021-08-10:
Initial take on this. The thinking is distinguish between simulation and upload on the main interface of the robot, so users can see something distinct.

On the client, they can see upload state via post response, and for simulation/validation, clients should be able to query status of the robot which can include simulation status in the response.

[image]

emilywools, 2021-08-10:
i think that makes sense, especially if the App is reinforcing this by having a consistent message with it.

readywater, 2021-09-20:
Progress:
- Upload Protocol
- Get back status for simulation
- Get back Notification on finished
- Poll protocol resource (pending to complete w/ results)

No limit on how many these can be done, eg. multiple sessions

mcous, 2021-11-22:
Closing this ticket out! PUR work has resulted in protocol analyses running in the background, with queryable status for the client

closed 2021-11-22 by mcous

emilywools, 2021-11-22:
whoa so we are going to be implementing this for 5.0???

mcous, 2021-11-22:
> whoa so we are going to be implementing this for 5.0???

What do you mean by "this"? I was just doing some quick cleanup of Andrew's tickets because the description seemed out of date with what the new PUR HTTP API already does:

- Upon protocol.uplod, analysis proceeds in the background
- Analysis will complete at some future point, and the client may query the analysis' status
- Running a protocol analysis does not "lock up" the robot, multiple analyses may be running concurrently with any other process

My bad if this ticket represented something else! Please re-open if I was mistaken

emilywools, 2021-11-22:
@mcous i was referring to the LED button on the robot