Opentrons/opentrons #8165: bug: The set protocol designer engage height doesn't match the actual height

Gyliann, 2021-07-26:
# Overview

When a user sets the magnetic module engage eight in Protocol Designer, it doesn't match the actual magnetic module engage height  

# Steps to reproduce

Create a new protocol with a magnet step in Protocol Designer
Set the magnetic action to engage and set the magnet to a particular height 

# Current behavior
The magnet engagement height set doesn't match the actual magnet engagement height

# Expected behavior

The magnet engagement height set matches the actual magnet engagement height

SyntaxColoring, 2021-07-26:
Is this with a GEN1 or GEN2 Magnetic Module? GEN1 Magnetic Modules don't use a standard unit of measurement for their engage height. GEN2 Magnetic Modules should be in millimeters.

Gyliann, 2021-07-26:
GEN2 Magnetic Module

alexjoel42, 2021-07-26:
Hi there! 

I just wanted to make sure that we had some data so this was easy to reproduce. 

I have two protocols below with the same:

1. Steps (engage and disengage)
2. Labware (NEST 96 Well Plate 100 µL PCR Full Skirt) 
3. GEN2 Magnetic Module 

But the engagement heights in the serial.log file appear to be different. 

Python has the engagement height at G0 Z8.5, Protocol Designer has the engagement height at b'G0 Z10.0. I attached the logs, but wanted to describe the reference points below. 

## Python: 
# Api.log 
```
Jul 26 21:57:39 friendship opentrons-api[242]: MagneticModuleContext.engage: {'height': None, 'offset': None, 'height_from_base': 6}
Jul 26 21:57:39 friendship opentrons-api[242]: MagneticModuleContext.disengage: {}
```
# serial.log 
```
Jul 26 21:58:38 friendship opentrons-api-serial[242]: magdeck 1930819568: Write -> b'G0 Z8.5 \r\n\r\n'

Jul 26 21:58:41 friendship opentrons-api-serial[242]: magdeck 1930819568: Write -> b'G0 Z0.0  \r\n\r\n'
```

## Protocol Designer: 

# api 
```
Jul 26 21:58:38 friendship opentrons-api[242]: MagneticModuleContext.engage: {'height': None, 'offset': None, 'height_from_base': 6}
Jul 26 21:58:38 friendship opentrons-api[242]: MagneticModuleContext.disengage: {}

```
# serial.log 
```
Jul 26 21:56:51 friendship opentrons-api-serial[242]: magdeck 1930819568: Write -> b'G0 **Z10.0** \r\n\r\n'
Jul 26 21:56:54 friendship opentrons-api-serial[242]: magdeck 1930819568: Write -> b'G0 **Z0.0** \r\n\r\n'
```
[api.log](https://github.com/Opentrons/opentrons/files/6881513/api.log)
[serial.log](https://github.com/Opentrons/opentrons/files/6881514/serial.log)

```python3

from opentrons import protocol_api

metadata = {'apiLevel': '2.10'}

def run(protocol: protocol_api.ProtocolContext):
    mag_mod = protocol.load_module('magnetic module gen2', '1')
    plate = mag_mod.load_labware('nest_96_wellplate_100ul_pcr_full_skirt')
    mag_mod.engage()
    mag_mod.disengage()
```

[testing_magnetic_module_for_bug.txt](https://github.com/Opentrons/opentrons/files/6881440/testing_magnetic_module_for_bug.txt).   (This can be converted easily back to json, it just wouldn't upload).

IanLondon, 2021-07-29:
This seems to be a problem with the Python API. I just verified that PD is saving expected commands.

The JSON executor in Python does `module.engage(height_from_base=engage_height)`

Looking at `module_context.py`, I see:

```
           - The recommended way to adjust the height of the magnets is to
             specify ``height_from_base``, which should be a distance in mm
             relative to the base of the labware that is on the magnetic module
           - If ``height`` is specified, it should be a distance in mm from the
             home position of the magnets.
```

@Gyliann @alexjoel42 could the discrepancy be that in the Python protocol you're comparing, it's using `engage(my_height)` while a JSON protocol would do `engage(height_from_base=my_height)`?

SyntaxColoring, 2022-02-19:
I'm closing this in favor of #9529. @Gyliann and @alexjoel42's observations are correct, but I think some things got flipped around between the Python and Protocol Designer test cases, and I thought it would be clearer to start with a clean slate.

Per @IanLondon's comment above:

> I just verified that PD is saving expected commands.

I'm not 100% confident that PD is correctly calculating default engage heights, because Python does it differently and the underlying schemas are ambiguous. I agree that it's saving the expected commands, but only if those default engage height calculations are correct.

> could the discrepancy be that in the Python protocol you're comparing, it's using engage(my_height) while a JSON protocol would do engage(height_from_base=my_height)?

Nope, it turns out that the problem is weirder. 🙃 

See [the new issue](https://github.com/Opentrons/opentrons/issues/9529) for details.