Opentrons/opentrons #8176: Python tests: Explicitly mark `async` functions instead of relying on `aiohttp` magic

SyntaxColoring, 2021-07-28:
Our Python tests define asynchronous test functions like this:

```python
async def test_my_thing() -> None:
    assert await foo() == bar
```

`pytest` doesn't natively support running `async` functions. It currently happens to magically work for us because we have the `pytest-aiohttp` plugin installed as a dev dependency.

For explicitness and normalcy, we instead want to:

* [ ] Add [`pytest-asyncio`](https://github.com/pytest-dev/pytest-asyncio) as a dev dependency.
* [ ] Explicitly mark our `async` test functions with `pytest.mark.asyncio`.
* [ ] Remove our dependency on `pytest-aiohttp` (assuming it's not used anywhere else.)

mcous, 2021-07-28:
AnyIO ships with a pytest plugin with the same `pytestmark` interface, if we wanted to add AnyIO as a dep or dev dep

SyntaxColoring, 2021-07-29:
Whichever library we choose, we should do some due diligence to see how it handles the fringes of setting up and tearing down async environments to run tests in. I only tenuously understand these concerns, but they seem surprisingly nuanced. Here are some questions I just opened on `pytest-asyncio`: https://github.com/pytest-dev/pytest-asyncio/issues/222

(But, also, it's not like we get to avoid these concerns by doing nothing and sticking with `pytest-aiohttp` magic.)

SyntaxColoring, 2022-04-16:
I tried implementing this with AnyIO's pytest plugin, but I found that its API has an unfortunate trap.

Briefly, if you have something like this:

```python
@pytest.fixture
async def my_async_fixture():
    ...

def my_non_async_test_func(fixture):
    ...
```

Then you'd expect `my_async_fixture()` to execute before `my_non_async_test_func()`, but in fact it never does.

This is because `my_non_async_test_func()` is, understandably, outside of the AnyIO plugin's consideration. It's just a normal non-async test function. So its fixture processing falls under pytest's normal rules. pytest "calls" `my_async_fixture()` like any other function. But because it's a coroutine, "calling" it
just returns an awaitable coroutine object; it doesn't actually run its code.

[Here's the specific fixture that tipped me off](https://github.com/Opentrons/opentrons/blob/64285130cc3b729dd981c9c31d9ffbf9133903a1/api/tests/opentrons/conftest.py#L106-L110), used by [this test](https://github.com/Opentrons/opentrons/blob/0e5191147652fbe8c0d50ccdf0e6f0ba3f33e0d9/api/tests/opentrons/hardware_control/test_pause_manager.py#L30-L50).

At best, this causes confusing errors, and at worst, it makes tests silently run under the wrong environment. After migrating around 90% of the `api` test suite and trying a couple workarounds to keep the tests passing, I consider it a deal-breaker. It makes things too fragile, and it's too surprising for the unwary. Especially for us in particular, since we frequently switch tests and fixtures between async and non-async, and we have a lot of fixtures living in `conftest.py` files far away from the things using them.

SyntaxColoring, 2024-01-02:
Closed by #9981, #10012, and #10022.

closed 2024-01-02 by SyntaxColoring