Opentrons/opentrons #8179: feat: Getting Started Examples Implemented in Jupyter Notebooks

readywater, 2021-07-28:
# Overview
The OT2 has a usable and high value Jupyter Notebook integration out of the box, but when a new OT2 user opens up this feature, they don't have a good place to start translating an analytically oriented python skillset (often the use case for those familiar with Jupyter Notebooks) to an automation oriented one.

Ideally, as a scientist who codes, I would want to have an easy set of example notebooks that would demonstrate how to get started with the OT-2. With clear examples and leveraging Jupyter Notebook's capacity for readable inline documentation, reproducibility, and modularity, the path to useful protocols will be faster.

# Implementation details

- Ideally translating existing and tested tutorial protocols vetted by support and user community
- Must highlight areas in documentation where Jupyter implementations differ/break relative to a "normal" python protocol format
- Should provide a skill ladder towards primary scientific applications, eg. PCR prep
- Should provide copyable snippets and opinionated implementations of common practices to tacitly demonstrate good "coding culture" to new OT2 users.

# Design

As a starting point, a few different examples could be used:

1. Hello world. Pick up a tip, then return it to its spot (examples for 1 channel and multichannel)
2. Dinosaur example implemented in Jupyter
3. Loading a CSV file to create a dynamic protocol
4. Adding a play/pause prompt for human interaction (eg. empty trash, replace tips, etc)
5. _Access the HTTP api and/or notification server_

# Acceptance criteria

### Design
- Clear and connected escalation in complexity across examples; successful testing of prototypes with a new OT2 user with prior Jupyter experience (but low to none non jupyter python experience)

### Development (app eng + cpx)
TODO

readywater, 2021-08-19:
TODO: Split into two tickets @readywater

SyntaxColoring, 2024-01-02:
I think this has been improved since this issue was written, but @ecormany FYI.

ecormany, 2024-01-02:
Yeah, the current state of docs isn't exactly all this. But we do have RTC-98 as a catch-all for "more better Jupyter", so closing this is fine.