Opentrons/opentrons #8185: RFC: maybe remove Codecov?

mcous, 2021-07-30:
## Overview

We started using Codecov to collect coverage statistics back in the heady days of #313. Those were simpler times, when we had an `api` project and an `app` project. Our previous coverage statistics provider, Coveralls, didn't support multiple languages at the time, hence the switch.

To say our Codecov coverage collection hasn't scaled gracefully would be putting it mildly. I think we have all collectively experienced the following:

- Nonsensical coverage comments on most PRs, including coverage changes for untouched files
- That uh, [huge security incident](https://www.reuters.com/technology/codecov-hackers-breached-hundreds-restricted-customer-sites-sources-2021-04-19/)
- Overall coverage statistics that don't even include all our projects, which was the original goal of codecov

Could we fix it? Maybe. Does anyone feel like it? I don't. **Should we get rid of Codecov?** Might be nice.

## Acceptance Criteria

- [ ] Remove codecov config and deactivate our repo in the Codecov UI
- [ ] Remove (currently wildly inaccurate) coverage badge from README.md

SyntaxColoring, 2021-07-30:
In theory, coverage comments on PRs sound really nice. If I add some new code, it's great to have a way of seeing if I forgot to cover some obscure execution path in that code with a test.

In practice, it's hard to trust Codecov's nonsensical coverage comments. 🙁 

Would things be better or worse if we applied Codecov to individual projects, instead of the whole repo? Is that even possible? I wonder if our fancy CI that avoids unnecessarily rebuilding unrelated subprojects is messing with our coverage uploads to Codecov, or something.

mcous, 2021-07-30:
[Codecov flags](https://docs.codecov.com/docs/flags) might be a potential solution here!

PR #8190 (2021-08-04): ci(codecov): fix codecov / jest coverage misconfigurations
## Overview

Upon some investigation for #8185, it turns out that the port from TravisCI to incremental builds in GitHub Actions never properly configured code coverage reporting for all of our JS projects, which explains our blatantly wrong coverage percentage in codecov.

This PR fixes coverage uploads and sets up per-project [Codecov flags](https://docs.codecov.com/docs/flags) to see if that makes a meaningful improvement to coverage comments in PRs.

Closes #4985, closes #8185

## Changelog

- Ensure unit-test CI workflows actually [... cut]
commit d515cc3d (2021-08-06): ci(codecov): fix codecov / jest coverage misconfigurations (#8190)

Closes #4985, closes #8185
closed 2021-08-04 by mcous