Opentrons/opentrons #8215: Labware Position Check: Intro Screen

b-cooper, 2021-08-09:
Implement the visuals detailed in this figma [node](https://www.figma.com/file/JCQNOTTK9tTmYPhRqeOfSj/Protocol-Upload-Revamp?node-id=4647%3A112043)

**Details**
- [ ] Fullscreen modal
- [ ] Title and description copy at the top of the modal
- [ ] Step summary with nothing selected on the left of the contents
- [ ] Deck map with all labware and modules on the right.
- [ ] Steps are highlighted with corresponding slots highlighted via an animation that takes 2 seconds for each step in order
- [ ] if user hovers over either a step or an occupied slot on the deck map, the animation is overridden and that step/slot highlight combo is momentarily fixed until the mouse leave the target.
- [ ] Call to action specifies the destination slot of the first command in the flow. Log a TODO onClick until further development.

mikec116, 2021-10-01:
Question from backlog grooming meeting: @shlokamin @Kadee80 Should we break out the last of these ACs into a new ticket and close this one?

shlokamin, 2021-10-01:
yeah that makes sense — @Kadee80 is working on a follow up right now. Katie would you mind adding the remaining ACs into a new ticket, deleting the new ones from this ticket, and then close this ticket afterwards?

commit 914f5cc1 (2021-10-11): refactor(app): update LPC description copy (#8493)

closes #8215