Opentrons/opentrons #8236: feat: more metadata info

splaisan, 2021-08-12:
It would eb cool to add more keys in the metadata section in order to document code versioning.
for instance add:

## more details than what "source" now reports
protocol_validation = "draft" "in development" "verified/validated" "production"

## info about the current code version
code_version or script_version = "1.00a"
.. script_date = "2021-08-12"
.. script_comment = "draft code, will be verified" or "verified / validated by John.Doe@lab.edu"

this kind of things present in all well written shared bioinfo scripts!

SyntaxColoring, 2021-08-13:
Thanks for the feature request!

To clarify, are you expecting these additional keys to show up in the Opentrons App somewhere, or do you just want the extra keys for seeing in a Python protocol's source code?

splaisan, 2021-08-14:
Mainly in the source code but seeing it on screen when running would not hurt either
Thanks

splaisan, 2021-09-16:
as discussed with Ethan, a similar metadata tag in the labware definition files would be essential as these also change upon time and you now may not know which version you are working with.

SyntaxColoring, 2021-09-21:
**Today:**

* You can include arbitrary custom fields in `metadata`. Currently, only strings are allowed.
   
   We haven't yet figured out a strategy for avoiding naming collisions between user-defined custom fields and official Opentrons fields. Until we figure that out, I strongly recommend giving your custom fields some unique prefix, like this:
       
   ```python
    metadata = {
        apiLevel = "2.11",
        myCompanyProtocolVersion = "1.00a",
        myCompanyProtocolValidation = "draft",
    }
    ```
* You can also, of course, record your custom information in a regular Python comment, like this:

    ```python
    # Protocol version: 1.00a
    # Protocol validation: draft
    
    metadata = {
        apiLevel = "2.11",
    }
    ```

    Or in some custom structure, like this:

    ```python
    my_company_metadata = {
        protocolVersion: "1.00a",
        protocolValidation: "draft",
    }

    metadata = {
        apiLevel = "2.11",
    }
    ```

**In the future:**

* We would like to formally specify what official metadata fields exist, to make it easier for protocol authors to use them correctly. See #8334. This will also include specifying a way for protocol authors to define their own custom metadata fields without worrying about naming collisions with our official ones.

splaisan, 2021-09-22:
Great implementation. I will use the proposed keys until more is announced.
Since it seems comment lines are not supported in json, how will we do the same in custom labware files?
Best regards and thanks for the nice work

splaisan, 2021-09-23:
OK, thank Max for the info

closed 2021-09-23 by splaisan